zephyrproject-rtos/zephyr #4607: tests/net/socket/udp/ is broken, again

pfalcon, 2017-10-27:
This feels like deja vu, because there was a ticket https://github.com/zephyrproject-rtos/zephyr/issues/4347 recently and the issue was fixed. My memory of what happened is:

1. Running "make run" in tests/net/socket/udp/ led to the test hang on start up (no testcase finished completely).
2. Then apparently, after #4347, it was fixed (i.e. test fully passed).

But now with the latest master I get:

~~~
qemu-system-i386: warning: Unknown firmware file in legacy mode: genroms/multiboot.bin

Running test suite socket_udp
===================================================================
starting test - test_send_recv_2_sock
PASS - test_send_recv_2_sock.
===================================================================
starting test - test_v4_sendto_recvfrom
~~~

And then it hangs.

But it's worse than that - trying to go to an earlier revision doesn't help, it all hangs in test_v4_sendto_recvfrom, until you reach fix for #4347, when it starts to hang at the beginning. And even checking out 7c60eadd5858b7242328aa2afbda346b4ce750f7, I still get hang in test_v4_sendto_recvfrom.

So, I then suspect my own system, but before rebooting, trying to shoot these tests into the CI via https://github.com/zephyrproject-rtos/zephyr/pull/4605 (that's why I started to run this tests in the first place now), and get the same behavior: https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/5943/3/tests (bottom).

@askawu : Do you have any ideas? Can you remember 100% clearly that you ran tests/net/socket/udp/ and all testcases worked for you?

pfalcon, 2017-10-27:
Even tried https://github.com/zephyrproject-rtos/zephyr/pull/4599, doesn't help.

pfalcon, 2017-10-28:
Well, the previous test in CI was not clean, here's updated log that confirms the problem: https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/5983/3/tests

askawu, 2017-10-30:
It's my fault.  It's the same problem of early version of #4390. These test cases are not self-contained. They should be run with the help of net-tools echo server. That's why it blocks the execution. I've submitted #4621 for the fix.

pfalcon, 2017-10-31:
Fixed, thanks.

closed 2017-10-31 by pfalcon