zephyrproject-rtos/zephyr #4630: Sample app 'coaps_server' fails to parse coap pkt

vakulgarg, 2017-10-30:
On running coaps_server sample, it fails and displays error:

"Could not parse packet".

I am using steps provided with example's README file.

jukkar, 2017-11-01:
This is related to issue described in commit 4fea5b18ab6aeb6db691589e227051a299f63528 
As there is no IP header after packet is received from DTLS tunnel, the coap library fails to parse it. I am working on a similar fix as described in above commit.

vakulgarg, 2017-11-02:
Hi Jukka

Isn't it a layer violation to require coap library having to deal with
transport (& possibly DTLS) headers?
IMO, coap library should be passed only the data relevant to it.
The other headers if any should be stripped by application itself.

Regards

Vakul

jukkar, 2017-11-02:
> Isn't it a layer violation to require coap library having to deal with transport (& possibly DTLS) headers? IMO, coap library should be passed only the data relevant to it. The other headers if any should be stripped by application itself.

Coap library does not need information about the actual data in transport headers, it needs to know where the user payload (=coap in this case) starts etc. So it might be enough if we set this accounting information properly in net_pkt, but I need to investigate this more. Currently the coaps code just strips away the transport headers but does not update the packet accounting information.

PR #4686 (2017-11-02): samples: net: coaps_server: Fix coap packet parsing
The length and the start of the coap payload was not correct
because the received packet did not had IP and UDP header in place.

Fixes #4630
commit 862ddb17 (2017-11-02): samples: net: coaps_server: Fix coap packet parsing

The length and the start of the coap payload was not correct
because the received packet did not had IP and UDP header in place.

Fixes #4630
commit 97c0aed6 (2017-11-04): samples: net: coaps_server: Fix coap packet parsing

The length and the start of the coap payload was not correct
because the received packet did not had IP and UDP header in place.

Fixes #4630
closed 2017-11-02 by jukkar