zephyrproject-rtos/zephyr #4634: LwM2M: no tests to ensure working functionality

mike-scott, 2017-10-30:
Need to discuss and implement a sane approach for running functional tests on the LwM2M subsystem.

mike-scott, 2019-08-01:
per @jukkar:
> Not necessarily part of this PR, but I just noticed that we are missing lwm2m settings from tests/net/all/prj.conf file. The file tries to enable max number of options from networking subsystem. It would be great to have lwm2m there too. There is tests/net/all/check_net_options.sh that is currently missing lwm2m checks and can be used to generate a list of currently missing options from prj.conf file. As I said, no need to add this support in this PR (so that we get this one merged sooner than later) but it would be nice to have lwm2m in tests/net/all at some point.

mike-scott, 2019-08-01:
I'll try and target adding LwM2M to check_net_options for v2.1

SeppoTakalo, 2023-10-12:
I propose that this issue is closed.

We have:
* `tests/net/lib/lwm2m/engine` Module test for observations
* `tests/net/lib/lwm2m/interop` Interoperability tests against Leshan server
* `tests/net/lib/lwm2m/lwm2m_engine` Module test for socket engine
* `tests/net/lib/lwm2m/lwm2m_rd_client` Module test for registration state machine
* `tests/net/lib/lwm2m/lwm2m_registry` Module test for resource handling

rlubos, 2023-10-12:
I agree 👍

closed 2023-10-12 by rlubos