zephyrproject-rtos/zephyr #4638: Shippable is failing when newlibc is enbaled

ramakrishnapallala, 2017-10-31:
timer_monotonic (tests/kernel/timer/timer_monotonic/) is throwing following warnings when we compile the test with newlibc enabled (CONFIG_NEWLIB_LIBC=y).

And due this shippable is failing on newlibc dependent patches.

In file included from /home/rpallala/ssg-otc/zephyr/31st-Oct/zephyr/tests/kernel/timer/timer_monotonic/src/main.c:9:0:
/home/rpallala/ssg-otc/zephyr/31st-Oct/zephyr/tests/kernel/timer/timer_monotonic/src/main.c: In function âmainâ:
/home/rpallala/ssg-otc/zephyr/31st-Oct/zephyr/tests/kernel/timer/timer_monotonic/src/main.c:60:13: warning: format â%ldâ expects argument of type âlong intâ, but argument 2 has type âs32_t {aka int}â [-Wformat=]
    TC_PRINT("diff = %" PRId32 " (t_last = %" PRIu32
             ^
/home/rpallala/ssg-otc/zephyr/31st-Oct/zephyr/tests/include/tc_util.h:22:37: note: in definition of macro âPRINT_DATAâ
 #define PRINT_DATA(fmt, ...) printk(fmt, ##__VA_ARGS__)
                                     ^~~
/home/rpallala/ssg-otc/zephyr/31st-Oct/zephyr/tests/kernel/timer/timer_monotonic/src/main.c:60:4: note: in expansion of macro âTC_PRINTâ
    TC_PRINT("diff = %" PRId32 " (t_last = %" PRIu32
    ^~~~~~~~
/home/rpallala/ssg-otc/zephyr/31st-Oct/zephyr/tests/kernel/timer/timer_monotonic/src/main.c:60:13: warning: format â%luâ expects argument of type âlong unsigned intâ, but argument 3 has type âu32_t {aka unsigned int}â [-Wformat=]
    TC_PRINT("diff = %" PRId32 " (t_last = %" PRIu32
             ^
/home/rpallala/ssg-otc/zephyr/31st-Oct/zephyr/tests/include/tc_util.h:22:37: note: in definition of macro âPRINT_DATAâ
 #define PRINT_DATA(fmt, ...) printk(fmt, ##__VA_ARGS__)
                                     ^~~
/home/rpallala/ssg-otc/zephyr/31st-Oct/zephyr/tests/kernel/timer/timer_monotonic/src/main.c:60:4: note: in expansion of macro âTC_PRINTâ
    TC_PRINT("diff = %" PRId32 " (t_last = %" PRIu32
    ^~~~~~~~
/home/rpallala/ssg-otc/zephyr/31st-Oct/zephyr/tests/kernel/timer/timer_monotonic/src/main.c:60:13: warning: format â%luâ expects argument of type âlong unsigned intâ, but argument 4 has type âu32_t {aka unsigned int}â [-Wformat=]
    TC_PRINT("diff = %" PRId32 " (t_last = %" PRIu32
             ^
/home/rpallala/ssg-otc/zephyr/31st-Oct/zephyr/tests/include/tc_util.h:22:37: note: in definition of macro âPRINT_DATAâ
 #define PRINT_DATA(fmt, ...) printk(fmt, ##__VA_ARGS__)
                                     ^~~
/home/rpallala/ssg-otc/zephyr/31st-Oct/zephyr/tests/kernel/timer/timer_monotonic/src/main.c:60:4: note: in expansion of macro âTC_PRINTâ
    TC_PRINT("diff = %" PRId32 " (t_last = %" PRIu32
    ^~~~~~~~

ramakrishnapallala, 2017-10-31:
The issue coming from the following TC_PRINT statement:
tests/kernel/timer/timer_monotonic/src/main.c

TC_PRINT("diff = %" PRId32 " (t_last = %" PRIu32
                                 " : t_now = %" PRIu32 "); i = %u\n",

If change the statement as below, no issue is seen:
 TC_PRINT("diff = %d (t_last = %u : t_now = %u); i = %u\n",
                                 diff, t_last, t_now, i);

what is the significance of using "PRId32/PRIu32" ?

andrewboie, 2017-11-01:
The fix is to not mix zephyr/types.h and newlib's inttypes.h which this code appears to be doing

andrewboie, 2017-11-01:
> what is the significance of using "PRId32/PRIu32" ?

http://pubs.opengroup.org/onlinepubs/009695399/basedefs/inttypes.h.html

We used to use these.
Then it was decided for Zephyr to use its own types instead.

ramakrishnapallala, 2017-11-02:
Can i the "inttypes.h" and use regular format specifiers(%d/, %u) to fix the issue?

ramakrishnapallala, 2017-11-06:
PR merged:
https://github.com/zephyrproject-rtos/zephyr/pull/4703

closed 2017-11-06 by ramakrishnapallala