zephyrproject-rtos/zephyr #4644: Kconfig warnings when building any sample for nRF5x

carlescufi, 2017-10-31:
```
warning: (SOC_SERIES_NRF51X && SOC_SERIES_NRF52X && GPIO_DW_CLOCK_GATE && SPI_DW_CLOCK_GATE) selects CLOCK_CONTROL which has unmet direct dependencies (SOC_SERIES_KINETIS_K6X && SOC_MK64F12 || SOC_SERIES_KINETIS_KL2X && SOC_MKL25Z4 || SO
C_SERIES_KINETIS_KWX && (SOC_MKW22D5 || SOC_MKW24D5) || SOC_SERIES_KINETIS_KWX && SOC_MKW40Z4 || SOC_SERIES_KINETIS_KWX && SOC_MKW41Z4)
warning: (SOC_SERIES_NRF51X && SOC_SERIES_NRF52X && GPIO_DW_CLOCK_GATE && SPI_DW_CLOCK_GATE) selects CLOCK_CONTROL which has unmet direct dependencies (SOC_SERIES_KINETIS_K6X && SOC_MK64F12 || SOC_SERIES_KINETIS_KL2X && SOC_MKL25Z4 || SO
C_SERIES_KINETIS_KWX && (SOC_MKW22D5 || SOC_MKW24D5) || SOC_SERIES_KINETIS_KWX && SOC_MKW40Z4 || SOC_SERIES_KINETIS_KWX && SOC_MKW41Z4)
```

carlescufi, 2017-10-31:
Any idea what introduced this? I've seen it for the last few days

carlescufi, 2017-10-31:
@MaureenHelm seems like this commit by you is the culprit after bisecting: 7a60782f6767e22e502f5d24c8956de477c9b2fb

MaureenHelm, 2017-10-31:
This problem seems limited to Nordic boards, as I don't see it happening on NXP or ST boards. Most of the clock control drivers are conditioned upon the SoC series, except for the NXP clock control driver which uses a `depends on HAS_MCUX_SIM` instead. I think the problem may be that the Nordic SoCs `select CLOCK_CONTROL` in Kconfig.series whereas everything else does it in defconfigs.

carlescufi, 2017-10-31:
@MaureenHelm thanks for looking into this. I tried selecting CLOCK_CONTROL in the board defconfig, but the problem is that our SoC requires clock control by default because it includes the NRF_TIMER, which replaces the SysTick and itself uses the clock control, so I don't think that's a solution, since clock control is an integral part of our SoCs and should indeed be in the Kconfig.series

MaureenHelm, 2017-10-31:
Can you do it in the SoC defconfig rather than the board defconfig? Although Kinetis doesn't use clock control to replace the SysTick, it does need clock control for the uart and thus the console. So it's pretty essential for Kinetis too.

carlescufi, 2017-11-01:
Addressed by: https://github.com/zephyrproject-rtos/zephyr/pull/4663

Build currently failing there because of a UART warning issue unrelated to this.

@MaureenHelm can you take a look?

carlescufi, 2017-11-01:
Fixed and merged by #4663

closed 2017-11-01 by carlescufi

MaureenHelm, 2017-11-01:
> @MaureenHelm can you take a look?

LGTM. Thanks!