zephyrproject-rtos/zephyr #4668: drivers/random/random_handlers.c is built when no random driver has been kconfig'ed into the build

SebastianBoe, 2017-11-01:
Hi,

after checking out the latest master I was surprised to see

   /home/sebo/zephyr-sdk-0.9.2-rc5/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elf/i586-zephyr-elf-ld -nostartfiles -nodefaultlibs -nostdlib -static   -r -o drivers/random/built-in.o drivers/random/random_handlers.o 

when building an application that didn't use a random driver. I am not familiar with the handler system,
but I assume the intention is to only build them when they are used by their driver.

This may apply to more handlers.

I need a confirmation whether this is a bug or not, to know if I should replicate the behaviour in cmake.

andrewboie, 2017-11-01:
I think the real problem is in drivers/Makefile

You'll see that most of the subsystems are only built if turned on by some Kconfig.
However, for console, entropy (random), serial, display, crc, and interrupt_controller, this is not the case.

In order to only build the entropy handlers if a entropy driver is built, need some Kconfig to indicate this condition.

andrewboie, 2017-11-01:
>  I assume the intention is to only build them when they are used by their driver.

They aren't used by the drivers per se.
They are the landing site when a user thread makes a system call to use some driver API. The handlers need to be built, otherwise any system call for that API will trigger an "unimplemented system call" error and kill the calling thread.
Their purpose is to validate the arguments passed in from user mode, and then call the real implementation.

SebastianBoe, 2017-11-01:
"In order to only build the entropy handlers if a entropy driver is built, need some Kconfig to indicate this condition."
Right. Long-term there should be some kind of standard, CONFIG_HAS_DRIVER_RANDOM, or whatever, you could probably find an ad-hoc standard by examining the KConfig sources.

Short-term you can close this as won't fix if you want. CMake is ifdef'ing away the entire random driver including the handler with the Kconfig option CONFIG_RANDOM_GENERATOR. If I understand you correctly it is desired behaviour that the handler is not present when the driver is not present.

andrewboie, 2017-11-13:
Nothing to do here since CMake does the right thing

closed 2017-11-13 by andrewboie