zephyrproject-rtos/zephyr #4682: http_server example fails for ESP32

micheljung, 2017-11-01:
OS: Windows 10
Board: ESP32
Zephyr at [e5b3918](https://github.com/zephyrproject-rtos/zephyr/commit/e5b3918a9ffc80e6ba079c6a0eff60aaa42917d0)
esp-idf at [dc8c338](https://github.com/espressif/esp-idf/commit/dc8c33892e0082822c4c69afa201c02d9758a1c0)
ESP32 Win32 msys2 environment: [2017-09-18](https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20170918.zip)

```
$ make BOARD=esp32 flash
make[1]: Entering directory '/s/sb-dev/zephyr'
make[2]: Entering directory '/s/sb-dev/zephyr/samples/net/http_server/outdir/esp32'
  CHK     include/generated/generated_dts_board.conf
  Using /s/sb-dev/zephyr as source for kernel
  GEN     ./Makefile
  CHK     include/generated/version.h
  CHK     include/generated/generated_dts_board.h
  CHK     include/generated/syscall_list.h
  CHK     include/generated/syscall_dispatch.c
  CHK     misc/generated/configs.c
  CHK     include/generated/offsets.h
  CC      lib/libc/minimal/source/stdlib/strtol.o
  CC      lib/libc/minimal/source/stdlib/strtoul.o
  CC      lib/libc/minimal/source/stdlib/atoi.o
  LD      lib/libc/minimal/source/stdlib/built-in.o
  CC      lib/libc/minimal/source/stdout/fprintf.o
  CC      lib/libc/minimal/source/stdout/prf.o
  CC      lib/libc/minimal/source/stdout/sprintf.o
  CC      lib/libc/minimal/source/stdout/stdout_console.o
  LD      lib/libc/minimal/source/stdout/built-in.o
  CC      lib/libc/minimal/source/string/string.o
  CC      lib/libc/minimal/source/string/strstr.o
  LD      lib/libc/minimal/source/string/built-in.o
  LD      lib/libc/minimal/source/built-in.o
  LD      lib/libc/minimal/built-in.o
  LD      lib/libc/built-in.o
  LD      lib/built-in.o
  CC      kernel/alert.o
  CC      kernel/device.o
  CC      kernel/errno.o
  CC      kernel/idle.o
  CC      kernel/init.o
  CC      kernel/mailbox.o
  CC      kernel/mem_slab.o
  CC      kernel/mempool.o
  CC      kernel/msg_q.o
  CC      kernel/mutex.o
  CC      kernel/pipes.o
  CC      kernel/poll.o
  CC      kernel/queue.o
  CC      kernel/sched.o
  CC      kernel/sem.o
  CC      kernel/stack.o
  CC      kernel/sys_clock.o
  CC      kernel/system_work_q.o
  CC      kernel/thread.o
  CC      kernel/thread_abort.o
  CC      kernel/timer.o
  CC      kernel/work_q.o
  AR      kernel/lib.a
  CC      drivers/console/uart_console.o
  LD      drivers/console/built-in.o
  CC      drivers/crc/crc16_sw.o
  LD      drivers/crc/built-in.o
  CC      drivers/gpio/gpio_esp32.o
  LD      drivers/gpio/built-in.o
  CC      drivers/i2c/i2c_esp32.o
  LD      drivers/i2c/built-in.o
  CC      drivers/pinmux/pinmux_esp32.o
  LD      drivers/pinmux/built-in.o
  CC      drivers/random/random_esp32.o
  LD      drivers/random/built-in.o
  CC      drivers/serial/uart_esp32.o
  LD      drivers/serial/built-in.o
  CC      drivers/timer/xtensa_sys_timer.o
  CC      drivers/timer/sys_clock_init.o
  LD      drivers/timer/built-in.o
  LD      drivers/built-in.o
  CC      misc/printk.o
  LD      misc/built-in.o
  CC      ext/lib/crypto/mbedtls/zephyr_init.o
  CC      ext/lib/crypto/mbedtls/library/aes.o
  CC      ext/lib/crypto/mbedtls/library/asn1parse.o
  CC      ext/lib/crypto/mbedtls/library/asn1write.o
  CC      ext/lib/crypto/mbedtls/library/base64.o
  CC      ext/lib/crypto/mbedtls/library/bignum.o
  CC      ext/lib/crypto/mbedtls/library/certs.o
  CC      ext/lib/crypto/mbedtls/library/cipher.o
  CC      ext/lib/crypto/mbedtls/library/cipher_wrap.o
  CC      ext/lib/crypto/mbedtls/library/ctr_drbg.o
  CC      ext/lib/crypto/mbedtls/library/des.o
  CC      ext/lib/crypto/mbedtls/library/entropy.o
  CC      ext/lib/crypto/mbedtls/library/entropy_poll.o
  CC      ext/lib/crypto/mbedtls/library/error.o
  CC      ext/lib/crypto/mbedtls/library/md.o
  CC      ext/lib/crypto/mbedtls/library/md5.o
  CC      ext/lib/crypto/mbedtls/library/md_wrap.o
  CC      ext/lib/crypto/mbedtls/library/memory_buffer_alloc.o
  CC      ext/lib/crypto/mbedtls/library/oid.o
  CC      ext/lib/crypto/mbedtls/library/pk.o
  CC      ext/lib/crypto/mbedtls/library/pk_wrap.o
  CC      ext/lib/crypto/mbedtls/library/pkparse.o
  CC      ext/lib/crypto/mbedtls/library/platform.o
  CC      ext/lib/crypto/mbedtls/library/rsa.o
  CC      ext/lib/crypto/mbedtls/library/sha1.o
  CC      ext/lib/crypto/mbedtls/library/sha256.o
  CC      ext/lib/crypto/mbedtls/library/ssl_ciphersuites.o
  CC      ext/lib/crypto/mbedtls/library/ssl_cli.o
  CC      ext/lib/crypto/mbedtls/library/ssl_srv.o
  CC      ext/lib/crypto/mbedtls/library/ssl_tls.o
  CC      ext/lib/crypto/mbedtls/library/timing.o
  CC      ext/lib/crypto/mbedtls/library/x509.o
  CC      ext/lib/crypto/mbedtls/library/x509_crt.o
  LD      ext/lib/crypto/mbedtls/built-in.o
  LD      ext/lib/crypto/built-in.o
  LD      ext/lib/built-in.o
  LD      ext/built-in.o
  CC      subsys/net/buf.o
  CC      subsys/net/ip/net_core.o
  CC      subsys/net/ip/net_if.o
  CC      subsys/net/ip/net_context.o
  CC      subsys/net/ip/net_pkt.o
  CC      subsys/net/ip/utils.o
  CC      subsys/net/ip/icmpv6.o
  CC      subsys/net/ip/nbr.o
  CC      subsys/net/ip/ipv6.o
  CC      subsys/net/ip/icmpv4.o
  CC      subsys/net/ip/ipv4.o
  CC      subsys/net/ip/route.o
  CC      subsys/net/ip/net_mgmt.o
  CC      subsys/net/ip/udp.o
  CC      subsys/net/ip/tcp.o
  CC      subsys/net/ip/net_shell.o
  CC      subsys/net/ip/net_stats.o
  CC      subsys/net/ip/connection.o
  LD      subsys/net/ip/built-in.o
  CC      subsys/net/lib/app/init.o
  LD      subsys/net/lib/app/built-in.o
  CC      subsys/net/lib/http/http_parser.o
  CC      subsys/net/lib/http/http_parser_url.o
  CC      subsys/net/lib/http/http_server.o
  LD      subsys/net/lib/http/built-in.o
  LD      subsys/net/lib/built-in.o
  LD      subsys/net/built-in.o
  CC      subsys/shell/shell_service.o
  CC      subsys/shell/shell.o
  LD      subsys/shell/built-in.o
  LD      subsys/built-in.o
  CC      arch/common/isr_tables.o
  LD      arch/common/built-in.o
  CC      arch/xtensa/core/cpu_idle.o
  CC      arch/xtensa/core/fatal.o
  CC      arch/xtensa/core/thread.o
  CC      arch/xtensa/core/xtensa_intr.o
  CC      arch/xtensa/core/irq_manage.o
  LD      arch/xtensa/core/built-in.o
  CC      arch/xtensa/soc/esp32/soc.o
  LD      arch/xtensa/soc/esp32/built-in.o
  LD      arch/xtensa/built-in.o
  LD      arch/built-in.o
  CC      src/main.o
  LD      src/built-in.o
  AR      libzephyr.a
  AR      libapplication.a
  LINK    zephyr.lnk
drivers/built-in.o: In function `esp32_uart_init':
S:/sb-dev/zephyr/drivers/serial/uart_esp32.c:46:(.text.esp32_uart_init+0x3): dangerous relocation: call8: call target out of range: esp32_rom_uart_attach
drivers/built-in.o: In function `esp32_uart_tx':
S:/sb-dev/zephyr/drivers/serial/uart_esp32.c:19:(.text.esp32_uart_tx+0x9): dangerous relocation: call8: call target out of range: esp32_rom_uart_tx_one_char
collect2.exe: error: ld returned 1 exit status
make[2]: *** [/s/sb-dev/zephyr/Makefile:876: zephyr_prebuilt.elf] Error 1
make[2]: Leaving directory '/s/sb-dev/zephyr/samples/net/http_server/outdir/esp32'
make[1]: *** [Makefile:178: sub-make] Error 2
make[1]: Leaving directory '/s/sb-dev/zephyr'
make: *** [S:/sb-dev/zephyr/Makefile.inc:88: flash] Error 2
```

lpereira, 2017-11-02:
It's not possible currently to use mbedtls on ESP32 due to the library size exceeding the few dozen kilobytes that the first stage bootloader will be able to load.

The first stage bootloader is part of the ESP32 ROM and has some limitations.  This bootloader currently loads everything into RAM and runs from there.  We would need a second stage bootloader that would load Zephyr from the flash and enable the flash cache, so that more code could be used.  I do have some code for this, but it's not working yet.

nashif, 2019-01-30:
http server removed

closed 2019-01-30 by nashif