zephyrproject-rtos/zephyr #4684: crypto apis not returning number of bytes written in output buffer

vakulgarg, 2017-11-02:
In structure cipher_pkt, there is an output field 'out_len'. This is supposed to return back to the caller number of bytes written in output buffer. 

The said field is left uninitialized by the crypto apis (e.g. cipher_ccm_op).

tbursztyka, 2017-11-02:
Up to the crypto driver to do so. But heh, now you mentioned this: both mtls and tc shim drivers do not set it. Only the ataes132c and cc2520 do it properly.

vakulgarg, 2017-11-02:
Right. Only mtls and tinycrypt shims miss it.

MaureenHelm, 2018-03-23:
@vakulgarg Is this still an issue?

vakulgarg, 2018-03-26:
Yes, it should still be an issue. I do not find any updates or fixes in the mtls, tinycrypt shim drivers.

MaureenHelm, 2018-04-13:
@tbursztyka @lpereira Will you please provide a fix?

lpereira, 2018-04-13:
@MaureenHelm Just sent a PR fixing this.

I'm not sure how correct this is, but it seems that, at least on the CCM encryption/decryption, both mbedTLS and TinyCrypt agree on how to calculate the output length.

For the other primitives supported by the TinyCrypt shim but not the mbedTLS shim, I believe things are correct after inspecting the respective TinyCrypt code, but I'd expect someone that knows more about this to review.

MaureenHelm, 2018-04-16:
@lpereira Thank you!

commit ab16853b (2018-05-16): drivers: crypto: crypto_mtls_shim: Set output length for all operations

The out_len member in struct cipher_pkt was not set.

Fixes #4684.
commit e61c4812 (2018-05-16): drivers: crypto: crypto_tc_shim: Set output length for all operations

The out_len member in struct cipher_pkt was not set.

Fixes #4684.
closed 2018-05-16 by nashif