zephyrproject-rtos/zephyr #4687: Windows: Remove all dependencies on MSYS2

SebastianBoe, 2017-11-02:
Definition Of Done: Windows users must be able to use all officially supported build system features without having a dependency on any form of unix emulation layer; MinGW/MSYS2/Cygwin/WSL.

Informally; Windows users should not need to install unix tools, the Zephyr host-side infrastructure should purely have portable dependencies such as python.

# Blockers
[ ] Kconfig
  [ ] merge_config.sh
  [ ] conf
  [ ] mconf
[ ] VM-VM networking
  [ ] named fifos

mbolivar, 2017-11-03:
Flashing and debugging bash scripts are done

andrewboie, 2017-11-03:
gperf might be a really tough one, it is used to generate perfect hash tables of kernel object pointers which are needed by memory protection features, if you don't want to use it we would have to re-implement it in Python.
@lpereira you were looking at this at one point?

lpereira, 2017-11-03:
@andrewboie I was.  I might return to it sometime if there's a real need.

I'm working on dtc right now.

lpereira, 2017-11-03:
A dtc implemented in Python is here: https://github.com/zephyrproject-rtos/zephyr/pull/4728

Some error checking should be introduced, but overall, this should work fine. Review appreciated. 

I'll look into gperf next week.  The prototype I have isn't working well in some cases; I might implement the same algorithm that gperf uses instead.

mbolivar, 2017-11-04:
> gperf might be a really tough one, it is used to generate perfect hash tables of kernel object pointers which are needed by memory protection features, if you don't want to use it we would have to re-implement it in Python.

The perfect hash table algorithm itself is not that complicated. What is the concern here that makes it a "really tough one"?

mbolivar, 2017-11-04:
Example: https://github.com/eddieantonio/perfection

lpereira, 2017-11-06:
There are many perfect hashes out there; we just need to find one that lets us build a hash function that do not require the use of a modulus operator (as integer division is quite costly).  Making the table size a power of two will help, but it's not always easy to do depending on the perfect hashing algorithm.  gperf isn't a perfect *minimal* hash table because of this.

But, yeah, I'm sure that if we shop around we'll find a suitable library.

nashif, 2017-12-12:
> merge_config.sh (?) (There is a minor bug in merge_config.py)

what is the bug? Can we open a new issue with more details, seems like a low hanging fruit :)

SebastianBoe, 2017-12-12:
Will do.

SebastianBoe, 2017-12-13:
merge_config now has a bugreport: https://github.com/zephyrproject-rtos/zephyr/issues/5374

carlescufi, 2018-01-23:
Closing this since Kconfig is now Python, gperf is a native Windows executable and we have an MSYS2-wrapped version of dtc that works reliably.

closed 2018-01-23 by carlescufi