zephyrproject-rtos/zephyr #4699: PWM Driver for ESP32

micheljung, 2017-11-02:
I'd like to use PWM with the ESP32, but (with my limited knowledge) I understand that there is no such driver yet? I expected it to be listed in #3471 but for some reason, it's not.

If I got it right and it's missing, I'm willing to pay a bounty for it.

lpereira, 2017-11-03:
What PWM controller are you talking about? There are two: LED control PWM and motor control PWM.

micheljung, 2017-11-03:
See, I didn't even know there's a difference ;) LED.

iha, 2017-11-06:
Hi @micheljung , I have started to study the ESP32 PWM in the datasheet and will keep you informed as soon as I have any progress.

iha, 2017-12-16:
@micheljung I made some tests and probably I'll send the PR until next Wednesday:

https://twitter.com/vitoriha/status/941654878862168064

lpereira, 2018-03-07:
Closing as merged.

closed 2018-03-07 by lpereira