zephyrproject-rtos/zephyr #4705: tests/net/socket/tcp/ :undefined reference to `__getreent

rrahulkx, 2017-11-03:
observed Build failure issue while running this testcase.

Work around:

testcase ---> zephyr/tests/net/socket/tcp

1) export ZEPHYR_GCC_VARIANT=espressif

2)make BOARD=esp32

Report is pasted below :---

xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib/libc.a(lib_a-strtol.o):(.literal+0x8): undefined reference to `__getreent'
xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib/libc.a(lib_a-strtol.o): In function `strtol':
/home/ivan/e/crosstool-NG/.build/src/newlib-2.2.0/newlib/libc/stdlib/strtol.c:223: undefined reference to `__getreent'
collect2: error: ld returned 1 exit status
/home/rrahulkx/zephyr/zephyr/Makefile:877: recipe for target 'zephyr_prebuilt.elf' failed
make[2]: *** [zephyr_prebuilt.elf] Error 1
make[2]: Leaving directory 'zephyr/tests/net/socket/tcp/outdir/esp32'
Makefile:178: recipe for target 'sub-make' failed

jukkar, 2017-11-03:
Actually this is not a networking issue but esp32 issue.

kumarvikash1, 2017-12-15:
@nashif , Who should be looking into this issue ?

lpereira, 2018-02-08:
Building with newlib is unsupported on ESP32.

nashif, 2018-02-09:
same issue as #1495

closed 2018-02-09 by nashif