zephyrproject-rtos/zephyr #4709: tests/kernel/fatal/testcase.yaml#stack-sentinel : Kernel Panic

rrahulkx, 2017-11-03:
When I ran this testcase .Their is failure due to kernel panic issues.

The report which I obtain is pasted below :--

Detailed description
====================
Automated test run failed.

Output (the target might have produced none; GDB will be needed at this point):

 build passed: 'mkdir -p outdir-qryy-lbab-quark_se_c1000_devboard; rm -f outdir-qryy-lbab-quark_se_c1000_devboard/.config'

 build passed: 'make -C zephyr/tests/kernel/fatal/ 
EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" 
KCPPFLAGS=-DTC_RUNID=:lbab 
BOARD=quark_se_c1000_devboard ARCH=x86  
O=outdir-qryy-lbab-quark_se_c1000_devboard initconfig'
 
build output: Using zephyr/boards/x86/quark_se_c1000_devboard/quark_se_c1000_devboard_defconfig as base

 build output: Merging prj.conf
 build output: Merging outdir-qryy-lbab-quark_se_c1000_devboard/500_boot_config.conf
 build output: Merging outdir-qryy-lbab-quark_se_c1000_devboard/500_bsp_arc_off.conf
 build output: # configuration written to .config
 
build passed: 'make -C zephyr/tests/kernel/fatal 
EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" 
KCPPFLAGS=-DTC_RUNID=:lbab 
BOARD=quark_se_c1000_devboard ARCH=x86 
 O=outdir-qryy-lbab-quark_se_c1000_devboard'
arc  build passed: 'mkdir -p outdir-qryy-lbab-quark_se_c1000_ss_devboard; rm -f outdir-qryy-lbab-quark_se_c1000_ss_devboard/.config'
 
build passed: 'make -C zephyr/tests/booting/stub/ 
EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations"
 KCPPFLAGS=-DTC_RUNID=:lbab 
BOARD=quark_se_c1000_ss_devboard ARCH=arc  
O=outdir-qryy-lbab-quark_se_c1000_ss_devboard initconfig'

build output: Merging zephyr/zephyr/tests/include/test.config

build output: Merging outdir-qryy-lbab-quark_se_c1000_ss_devboard/500_boot_config.conf
build output: # configuration written to .config

build passed: 'make -C zephyr/tests/booting/stub 

EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" 
KCPPFLAGS=-DTC_RUNID=:lbab 
BOARD=quark_se_c1000_ss_devboard ARCH=arc  
O=outdir-qryy-lbab-quark_se_c1000_ss_devboard'

build passed

deploy passed

Reset

console output: starting test - test_fatal
console output: test alt thread 1: generic CPU exception
console output: ***** CPU exception 6
console output: Current thread ID = 0xa8006420
console output: Faulting segment:address = 0x0008:0x40030157
console output: eax: 0x00000000, ebx: 0x40030154, ecx: 0x00000000, edx: 0x00000000
console output: esi: 0x400301cd, edi: 0x40031ae5, ebp: 0xa8006e38, esp: 0xa8006e38
console output: eflags: 0x10286
console output: Caught system error -- reason 6
console output: PASS
console output: test alt thread 2: initiate kernel oops
console output: ***** Kernel OOPS! *****
console output: Current thread ID = 0xa8006420
console output: Faulting segment:address = 0x0008:0x4003016f
console output: eax: 0x00000286, ebx: 0x40030165, ecx: 0x00000000, edx: 0x00000000
console output: esi: 0x40030008, edi: 0x40031ae5, ebp: 0xa8006e38, esp: 0xa8006e34
console output: eflags: 0x86
console output: Caught system error -- reason 7
console output: PASS
console output: test alt thread 3: initiate kernel panic
console output: ***** Kernel Panic! *****
console output: Current thread ID = 0xa8006420
console output: Faulting segment:address = 0x0008:0x40030179
console output: eax: 0x00000286, ebx: 0x4003016f, ecx: 0x00000000, edx: 0x00000000
console output: esi: 0x00000086, edi: 0x40031ae5, ebp: 0xa8006e38, esp: 0xa8006e34
console output: eflags: 0x86
console output: Caught system error -- reason 8
console output: PASS
console output: test stack overflow - timer irq
console output: posting 2048 bytes of junk to stack...
console output: busy waiting...
console output: FAIL - stack_thread1@106. should never see this
console output: FAIL - main@206. bad reason code got 8 expected 4
console output: FAIL - main@210. thread was not aborted
console output: FAIL - main.
console output:

andrewboie, 2017-11-03:
Why is this not failing in CI?
I think you are running this test with incorrect configuration.

nashif, 2018-02-09:
cant reproduce

closed 2018-02-09 by nashif