zephyrproject-rtos/zephyr #4724: sanitycheck build_only option can be confusing

nashif, 2017-11-03:
```
  build_only: <True|False> (default False)
    If true, don't try to run the test under QEMU even if the
    selected platform supports it.
```
A few things:
- this should not be a QEMU thing, although we do run most cases in qemu only, there might be other emulation platforms other than Qemu, for example xt-sim, renode, etc.
- We do have some testcases which are true 'build_only', meaning that should never run on anything, i.e not designed to run on hardware or emulation. For example we have footprint benchmarks that remove all kernel functionality with the sole purpose of measuring footprint.
- Many 'build_only' tests are marked as such to avoid running them in Qemu, however, they can run on real hardware without problems

the 'build_only' might be confusing  given the 3 cases above, so we should have some other way to mark tests that can run in emulation and hardware, run in emulation only, and finally those tests which are 'build_only'.

andrewboie, 2017-11-06:
> Many 'build_only' tests are marked as such to avoid running them in Qemu, however, they can run on real hardware without problems

I don't think this is a good way to look at it.
I think the QEMU targets should be treated like any other HW platform, with YAML files describing capabilities that are filtered on to run tests.
"build_only" should be used only for those tests that really shouldn't be run on any platform, for things  like footprint, or samples which do not report pass/fail status.
If people are using "build only" to just filter out QEMU because QEMU lacks some peripheral emulation or something, that is the wrong way to go about it.
I think the help test should be changed accordingly to remove all reference to any emulator and just say "don't run this just build it"

nashif, 2017-12-08:
> think the QEMU targets should be treated like any other HW platform, with YAML files describing capabilities that are filtered on to run tests.

Yes, that is the approach I was considering., maybe I was not clear

pfalcon, 2017-12-12:
There's another point to consider:

As discussed in https://github.com/zephyrproject-rtos/zephyr/pull/5349#issuecomment-351051653 , there can be also external test systems, which may be "dumb" (or rather, complex enough already, to not complicate them with additional processing/decision making), which would run any binary built (e.g., on real hardware, which isn't supported by sanitycheck). So, making sanitycheck smart regarding "build_only" isn't enough, "build_only" tests should be still treated as runnable tests (even if sanitycheck optimizes away their running).

PR #5410 (2017-12-29): Fix testcases and samples removing build_only tag
We should use build_only for tests that are not supposed to run on anything,
not even Qemu. Those build_only tests are just compile test to make sure some
code does not bitrot. Everything else that does not have this tag should be
runnable on some hardware or simulation.

To identify what sanitycheck should run in Qemu, we add depends_on where it
makes sense or whitelist some samples for specific platforms. If a sample or a
test can run in Qemu but requires some interaction or evaluation of the console
we use the harness keyword to specify what [... cut]
closed 2017-12-29 by nashif