zephyrproject-rtos/zephyr #4744: tests/net/ieee802154/l2/testcase.yaml#test : unable to acknowledge data from receiver

rrahulkx, 2017-11-06:
After re -producing this test case the failure log is pasted below.

**The Report is pasted below**:

configure passed
build passed: 'mkdir -p outdir-fm7j-rhok-qemu_x86; rm -f outdir-fm7j-rhok-qemu_x86/.config'

build passed: 'make -C zephyr/tests/net/ieee802154/l2/ 
EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" 
KCPPFLAGS=-DTC_RUNID=:rhok 
BOARD=qemu_x86 ARCH=x86   
O=outdir-fm7j-rhok-qemu_x86 initconfig'

build output: Merging prj.conf
build output: Merging outdir-fm7j-rhok-qemu_x86/500_boot_config.conf
build output: # configuration written to .config

build passed: 'make -C zephyr/tests/net/ieee802154/l2
 EXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" 
KCPPFLAGS=-DTC_RUNID=:rhok 
BOARD=qemu_x86 ARCH=x86   
O=outdir-fm7j-rhok-qemu_x86'

build passed

deploy passed

Reset

console output: Running test suite ieee802154_l2
console output: ===================================================================
console output: starting test - init_test
console output:  -> IEEE 802.15.4 Address: 00:12:4b:00:00:9e:a3:c2
console output: PASS - init_test.
console output: ===================================================================
console output: starting test - parsing_ns_pkt
console output: PASS - parsing_ns_pkt.
console output: ===================================================================
console output: starting test - sending_ns_pkt
console output:  -> Packet content:
console output:         41 d8 42 cd ab ff ff 00 00 00
console output:         00 00 00 00 00 7b 09 3a 20 01
console output:         db 08 00 00 00 00 00 00 00 00
console output:         00 00 00 01 02 01 ff 00 00 01
console output:         87 00 91 11 00 00 00 00 ff 02
console output:         00 00 00 00 00 00 00 00 00 01
console output:         ff 00 00 01 01
console output: PASS - sending_ns_pkt.
console output: ===================================================================
console output: starting test - parsing_ack_pkt
console output: PASS - parsing_ack_pkt.
console output: ===================================================================
console output: starting test - replying_ack_pkt
console output:  -> Packet content:
console output:         41 d8 43 cd ab ff ff 00 00 00
console output:         00 00 00 00 00 7b 3b 3a 01 85
console output:         00 99 41 01 ff 00 00 01
console output:
console output:     Assertion failed at /home/rrahulkx/zephyr/zephyr/tests/net/ieee802154/l2/src/ieee802154_test.c:335: replying_ack_pkt: (ret is false)
console output: ACK repliedFAIL - replying_ack_pkt.
console output: ===================================================================
console output: starting test - parsing_beacon_pkt
console output: PASS - parsing_beacon_pkt.
console output: ===================================================================
console output: starting test - parsing_sec_data_pkt
console output: PASS - parsing_sec_data_pkt.
console output: ===================================================================
console output: ===================================================================

tbursztyka, 2017-11-06:
Works just fine here, I have been running it many times.

Could you paste the generated .config ?

rrahulkx, 2017-11-24:
please find the .config  file attached generated for   **qemu -arm**.
[config_qemu_arm.txt](https://github.com/zephyrproject-rtos/zephyr/files/1500645/config_qemu_arm.txt)

tbursztyka, 2017-11-24:
Why qemu-arm? Your log above is about qemu-x86.

rrahulkx, 2017-11-24:
Because as per the Zephyr Commit ID:07e1a64e77e15180721c56993a19bce473ff4365

This TC failed on qemu_x86 and qemu_arm.

please find the config file for the qemu_x86
[config_qemu_x86.txt](https://github.com/zephyrproject-rtos/zephyr/files/1501414/config_qemu_x86.txt)

tbursztyka, 2017-11-24:
Ok. And it still works just fine here.
I run it the manual way:

cd tests/net/ieee802154/l2
mkdir build && cd build
cmake -DBOARD=qemu_x86 -DCONF_FILE=prj.conf ..
make run

Are you up-to-date on zephyr tree and sdk? (btw the commit ID you point has nothing to do with what's tested, actually I²C is not enabled for that test also).
I don't know what your environment is doing, there are slight changes in the .config

In "# Linker Options"
mine has: CONFIG_CHECK_LINK_MAP=y
yours don't

yours don't have the section "# Build Options"

That's the only difference.

rrahulkx, 2017-11-27:
ok  thanks @tbursztyka .
With latest zephyr(07e1a64e77e15180721c56993a19bce473ff4365), Testcase is passing, so closing the issue.

closed 2017-11-27 by rrahulkx