zephyrproject-rtos/zephyr #4759: [PTS] GATT/CL/GAW/BV-02-C fails with INCONC

nniranjhana, 2017-11-07:
_Test Env:_
Zephyr Revision: master/c968e102e679f41f0fd2016fea7391f4c121a320
PTS: 7.1.1
Host: Win 10 64 bit Enterprise, with Intel® Core i5-6300U CPU @2.40ghZ, 8 GB RAM
BT USB Dongle: PTS Dongle
Client: Arduino 101 flashed with current Zephyr Revision

nniranjhana, 2017-11-07:
[autoptsclient-zephyr.log](https://github.com/zephyrproject-rtos/zephyr/files/1448927/autoptsclient-zephyr.log)

MariuszSkamra, 2017-11-13:
This is PTS issue. We have reported this [here](https://bluetooth.service-now.com/x_bsig_case_manage_case.do?sys_id=c15fd23a4f8e8b407270b3318110c7ed&sysparm_view=ess&sysparm_record_target=x_bsig_case_manage_case&sysparm_record_row=2&sysparm_record_rows=4&sysparm_record_list=categoryANYTHING%5ErequesterDYNAMICjavascript%3Ags.getUserID%28%29%5Eaffect_layerANYTHING%5EORDERBYnumber). It is in resolved state now, you can test your solution with ETS attached there.

nniranjhana, 2017-11-13:
@MariuszSkamra Okay, thank you for the info. Do you have the link to the list of issues that have been reported such as this? I have few more, thought I could raise. Would like to check if they have already been reported.

nniranjhana, 2017-11-14:
@MariuszSkamra found the place, I will raise the other PTS issues where you have raised ones such as these before.

closed 2017-11-14 by nniranjhana