zephyrproject-rtos/zephyr #4760: stm32f4_disco samples/basic/blink_led ,Choose supported PWM driver

spring-z, 2017-11-07:
```
/Volumes/Embedded/Zephyr/zephyr/samples/basic/blink_led/src/main.c:33:2: error: #error "Choose supported PWM driver"
 #error "Choose supported PWM driver"
  ^
/Volumes/Embedded/Zephyr/zephyr/samples/basic/blink_led/src/main.c: In function 'main':
/Volumes/Embedded/Zephyr/zephyr/samples/basic/blink_led/src/main.c:50:31: error: 'PWM_DRIVER' undeclared (first use in this function)
  pwm_dev = device_get_binding(PWM_DRIVER);
                               ^
/Volumes/Embedded/Zephyr/zephyr/samples/basic/blink_led/src/main.c:50:31: note: each undeclared identifier is reported only once for each function it appears in
/Volumes/Embedded/Zephyr/zephyr/samples/basic/blink_led/src/main.c:57:33: error: 'PWM_CHANNEL' undeclared (first use in this function)
   if (pwm_pin_set_usec(pwm_dev, PWM_CHANNEL,
                                 ^
make[3]: *** [src/main.o] Error 1
make[2]: *** [zephyr-app-dir] Error 2
make[1]: *** [sub-make] Error 2
make: *** [all] Error 2

```

spring-z, 2017-11-08:
@galak

galak, 2017-11-08:
@spring-z can you try and explain what issue are you running into?

spring-z, 2017-11-08:
I try run samples to stm32f4_disco

galak, 2017-11-08:
What are you specific build steps?

spring-z, 2017-11-08:
➜  make BOARD=stm32f4_disco
Using /Volumes/Embedded/Zephyr/zephyr/boards/arm/stm32f4_disco/stm32f4_disco_defconfig as base
Merging prj.conf
#
# configuration written to .config
#
  GEN     ./Makefile
scripts/kconfig/conf --silentoldconfig Kconfig
  DTC     dts/arm/stm32f4_disco.dts_compiled
dts/arm/stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller has a reg or ranges property, but no unit name
dts/arm/stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart3@0 has a unit name, but no reg property
dts/arm/stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart3@1 has a unit name, but no reg property
dts/arm/stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart6@0 has a unit name, but no reg property
dts/arm/stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart6@1 has a unit name, but no reg property
dts/arm/stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@0 has a unit name, but no reg property
dts/arm/stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@1 has a unit name, but no reg property
dts/arm/stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@2 has a unit name, but no reg property
dts/arm/stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@0 has a unit name, but no reg property
dts/arm/stm32f4_disco.dts_compiled: Warning (simple_bus_reg): Node /soc/pin-controller simple-bus unit address format error, expected "40020000"
dts/arm/stm32f4_disco.dts_compiled: Warning (simple_bus_reg): Node /soc/i2c@40005C00 simple-bus unit address format error, expected "40005c00"
dts/arm/stm32f4_disco.dts_compiled: Warning (clocks_property): Missing property '#clock-cells' in node /soc/rcc@40023800 or bad phandle (referred from /soc/serial@40011000:clocks[0])
dts/arm/stm32f4_disco.dts_compiled: Warning (clocks_property): Missing property '#clock-cells' in node /soc/rcc@40023800 or bad phandle (referred from /soc/serial@40004400:clocks[0])
dts/arm/stm32f4_disco.dts_compiled: Warning (clocks_property): Missing property '#clock-cells' in node /soc/rcc@40023800 or bad phandle (referred from /soc/serial@40011400:clocks[0])
dts/arm/stm32f4_disco.dts_compiled: Warning (clocks_property): Missing property '#clock-cells' in node /soc/rcc@40023800 or bad phandle (referred from /soc/serial@40004800:clocks[0])
dts/arm/stm32f4_disco.dts_compiled: Warning (clocks_property): Missing property '#clock-cells' in node /soc/rcc@40023800 or bad phandle (referred from /soc/serial@40004c00:clocks[0])
dts/arm/stm32f4_disco.dts_compiled: Warning (clocks_property): Missing property '#clock-cells' in node /soc/rcc@40023800 or bad phandle (referred from /soc/serial@40005000:clocks[0])
  CHK     include/generated/generated_dts_board.conf
  UPD     include/generated/generated_dts_board.conf
  CHK     include/generated/generated_dts_board.conf
  Using /Volumes/Embedded/Zephyr/zephyr as source for kernel
  GEN     ./Makefile
  CHK     include/generated/version.h
  UPD     include/generated/version.h
  CHK     include/generated/generated_dts_board.h
  UPD     include/generated/generated_dts_board.h
  CHK     include/generated/syscall_list.h
  UPD     include/generated/syscall_list.h
  CHK     include/generated/syscall_dispatch.c
  UPD     include/generated/syscall_dispatch.c
  CHK     misc/generated/configs.c
  UPD     misc/generated/configs.c
  CHK     include/generated/offsets.h
  UPD     include/generated/offsets.h
  CC      lib/libc/minimal/source/stdlib/strtol.o
  CC      lib/libc/minimal/source/stdlib/strtoul.o
  CC      lib/libc/minimal/source/stdlib/atoi.o
  LD      lib/libc/minimal/source/stdlib/built-in.o
  CC      lib/libc/minimal/source/stdout/fprintf.o
  CC      lib/libc/minimal/source/stdout/prf.o
  CC      lib/libc/minimal/source/stdout/sprintf.o
  CC      lib/libc/minimal/source/stdout/stdout_console.o
  LD      lib/libc/minimal/source/stdout/built-in.o
  CC      lib/libc/minimal/source/string/string.o
  CC      lib/libc/minimal/source/string/strncasecmp.o
  CC      lib/libc/minimal/source/string/strstr.o
  LD      lib/libc/minimal/source/string/built-in.o
  LD      lib/libc/minimal/source/built-in.o
  LD      lib/libc/minimal/built-in.o
  LD      lib/libc/built-in.o
  LD      lib/built-in.o
  LD      tests/built-in.o
  CC      kernel/version.o
  LD      kernel/built-in.o
  CC      kernel/alert.o
  CC      kernel/device.o
  CC      kernel/errno.o
  CC      kernel/idle.o
  CC      kernel/init.o
  CC      kernel/mailbox.o
  CC      kernel/mem_slab.o
[... 85 lines cut ...]
  LD      arch/arm/soc/st_stm32/common/built-in.o
  CC      arch/arm/soc/st_stm32/stm32f4/soc.o
  CC      arch/arm/soc/st_stm32/stm32f4/soc_gpio.o
  LD      arch/arm/soc/st_stm32/stm32f4/built-in.o
  LD      arch/arm/soc/st_stm32/built-in.o
  LD      arch/arm/built-in.o
  CC      arch/common/isr_tables.o
  LD      arch/common/built-in.o
  LD      arch/built-in.o
  CC      src/main.o
/Volumes/Embedded/Zephyr/zephyr/samples/basic/blink_led/src/main.c:33:2: error: #error "Choose supported PWM driver"
 #error "Choose supported PWM driver"
  ^
/Volumes/Embedded/Zephyr/zephyr/samples/basic/blink_led/src/main.c: In function 'main':
[... cut ...]
/Volumes/Embedded/Zephyr/zephyr/samples/basic/blink_led/src/main.c:50:31: note: each undeclared identifier is reported only once for each function it appears in
/Volumes/Embedded/Zephyr/zephyr/samples/basic/blink_led/src/main.c:57:33: error: 'PWM_CHANNEL' undeclared (first use in this function)
   if (pwm_pin_set_usec(pwm_dev, PWM_CHANNEL,
                                 ^
make[3]: *** [src/main.o] Error 1
make[2]: *** [zephyr-app-dir] Error 2
make[1]: *** [sub-make] Error 2
make: *** [all] Error 2

galak, 2017-11-08:
Based on zephyr/samples/basic/blink_led/sample.yaml the only platforms this sample is buildable for are:
      platform_whitelist: arduino_101 quark_d2000_crb
        nucleo_f103rb nucleo_f401re nucleo_l476rg hexiwear_k64

So one needs to port the sample for the stm32f4_disco board

spring-z, 2017-11-08:
Thanks, I need STM32 samples

spring-z, 2017-11-08:
add PR

spring-z, 2017-11-08:
@erwango

spring-z, 2017-11-08:
Sorry. CI block. I am trying to meet the standard.

PR #4824 (2017-11-14): samples: basic: blink_led: add support PWM for the stm32f4_disco
Fix #4760
closed 2017-11-14 by nashif