zephyrproject-rtos/zephyr #4779: TCP server doesn't send FIN when performing active close.

askawu, 2017-11-08:
Hi, 

I found TCP server doesn't send FIN to the client when performing active close.  Following are how to reproduce this issue.

### Zepher as TCP server
- Apply patches
```
diff --git a/samples/net/sockets/echo/prj.conf b/samples/net/sockets/echo/prj.conf
index 58eb89c..2d9cd49 100644
--- a/samples/net/sockets/echo/prj.conf
+++ b/samples/net/sockets/echo/prj.conf
@@ -23,4 +23,7 @@ CONFIG_NET_APP_PEER_IPV4_ADDR="192.0.2.2"

 # Network debug config
 #CONFIG_NET_DEBUG_SOCKETS=y
-CONFIG_SYS_LOG_NET_LEVEL=2
+CONFIG_NET_LOG=y
+CONFIG_SYS_LOG_NET_LEVEL=4
+CONFIG_NET_DEBUG_CONTEXT=y
+CONFIG_NET_DEBUG_CORE=y
```
```
diff --git a/samples/net/sockets/echo/src/socket_echo.c b/samples/net/sockets/echo/src/socket_echo.c
index 52291ab..d7bcd54 100644
--- a/samples/net/sockets/echo/src/socket_echo.c
+++ b/samples/net/sockets/echo/src/socket_echo.c
@@ -53,6 +53,13 @@ int main(void)
                        if (len == 0) {
                                break;
                        }
+
+                       if (memcmp(buf, "bye", 3) == 0) {
+                               printf("Active close connection !!\n");
+                               close(serv);
+                               return 0;
+                       }
+
                        send(client, buf, len, 0);
                }

```
- cd samples/net/sockets/echo/; make run

### Netcat as TCP client
- echo "bye" > msg
- cat bye - | nc 192.0.2.1 4242
   - "cat" will wait for stdin, so the connection will not be closed by netcat immediately.

### Zephyr Log
```
[net/core] [DBG] pr[net/core] [DBG] net_recv_data: (0x00401060): fifo 0x00401048 iface 0x004192e0 pkt 0x004032f0 len 58
ocessing_data: (0x00401060): Dropping pkt 0x0040332c
[net/core] [DBG] net_rx_thread: (0x00401060): Received pkt 0x004032f0 len 58
[net/ctx] [DBG] _tcp_established: (0x00401060): DATA received from 192.0.2.2 port 51920
[net/ctx] [DBG] packet_received: (0x00401060): Set appdata 0x004029d6 to len 4 (total 44)
[net/ctx] [DBG] print_send_info: (0x00401060): ACK sent to 192.0.2.2 port 51920
[net/core] [DBG] processing_data: (0x00401060): Consumed pkt 0x004032f0
Connection #0 from 192.0.2.2
Active close connection !!
[net/ctx] [DBG] net_context_put: (0x00401a20): TCP connection in active close, not disposing yet (waiting 1000ms)
[net/core] [DBG] check_ip_addr: (0x00401a20): IPv4 dst address missing
[net/ctx] [DBG] handle_fin_timeout: (0x00401bd0): Did not receive FIN in 1000ms
[net/ctx] [DBG] net_context_unref: (0x00401bd0): Context 0x004011a0 released
```

### Summary
- In net_context_put(), queue_fin() will be called to send FIN.
- In queue_fin(), FIN is generated by net_tcp_prepare_segment() and the remote address of FIN comes from the remote address of the net context.
- The pkt will be sent by calling net_send_data(). However, it fails at check_ip_addr() because the remote address is unspecified, i.e zero.
- For a listening net context, the connection will be something like "**[0/10/6/0x05] remote 0x00000000/-/0 local 0x00412290/::/4242**", the remote address is never assigned.

### Expectation
- When TCP server performs active close, FIN should be sent to all the connections with the same local address and local port.

pfalcon, 2018-01-26:
@askawu : Any chance that you can re-test this, I believe there were some fixes regarding close behavior.

jukkar, 2018-02-19:
@rveerama1 has fixed TCP issues lately so this should be fixed already and requires re-testing.

laperie, 2018-03-06:
@jukkar , @rveerama1  -- shall we close it, do we have any TC that is close to the original descrption?

jukkar, 2018-03-06:
Lets close this one, we can re-open of course if the problem persists.

closed 2018-03-06 by jukkar