zephyrproject-rtos/zephyr #4836: connection disconnected due to Le timeout(0x22)

cobbler123, 2017-11-09:
Hi friend,
             When I test some android phone (such as 360 N5,xiaomi3,xiaomi4C,MotoX) connecting with My BLE Evb. The connection will be failed due to le timeout(0x22).
             After some debug and  analyse,  I get some clues about it:
             (1) this issue is related with feature –“LE_SLAVE_FEAT_REQ”. When I don’t call hci_le_read_remote_features, It is ok!
             (2) when I adding 1s delay before calling the function, It also Ok!
             
            I capture sniffer log, please help me analyzing this issue.
           PS: attachment, please change *.txt to *.rar .

Thanks

[CordioZephyrfailed.txt](https://github.com/zephyrproject-rtos/zephyr/files/1457337/CordioZephyrfailed.txt)

jhedberg, 2018-02-08:
@cobbler123 are you still able to reproduce this with the latest master branch?

galak, 2018-12-13:
Due to lack of response going to close, feel free to re-open if needed

closed 2018-12-13 by galak