zephyrproject-rtos/zephyr #4851: cmake does not show verbose output of build tools

andrewboie, 2017-11-09:
In the old build system, if V=1 was passed to Make, then this would implictly pass --verbose flags to tools like gen_idt, gen_gdt, gen_kobject_list, etc...
This isn't happening any more, if I pass VERBOSE=1 to the build I see no debug output from post build tools.

nashif, 2017-11-09:
Does 

`cmake -DCMAKE_VERBOSE_MAKEFILE=ON .. -DBOARD=qemu_x86`

give you what you want?

andrewboie, 2017-11-09:
that prints the output I am expecting yes
can we make "make VERBOSE=1" do this??

SebastianBoe, 2017-11-10:
I'm not able to reproduce what was originally reported here.

Steps to reproduce:
https://asciinema.org/a/pseZ5zugMwdyaJ3ynCDQvnVUp

SebastianBoe, 2017-11-10:
I can't see an easy way to do that unfortunately.

carlescufi, 2017-11-24:
@andrewboie try `make --trace`

SebastianBoe, 2017-11-24:
That won't work Carles. The build system needs to invoke the underlying commands differently.

Make does not know that our custom commands accept a "--verbose" flag.

The only solution I can see here is having the scripts accept environment variables.

pfalcon, 2017-11-24:
For me, `make VERBOSE=1` usually works (without having added any special flags to cmake previously). But sometimes it doesn't, pattern is not clear so far.

SebastianBoe, 2017-11-24:
The desired design is that "make VERBOSE=1" is used.

But today, if you want a truly verbose build, you need to do

cmake -DCMAKE_VERBOSE_MAKEFILE=ON && make VERBOSE=1

pfalcon, 2017-11-24:
Attempt to established a more general outlook of the problems we experience with make variables: https://github.com/zephyrproject-rtos/zephyr/issues/5153

PR #5503 (2018-01-11): cmake: Fix "make VERBOSE=1"
It is desired behaviour that when 'make VERBOSE=1' is invoked, the
underlying scripts will be invoked in 'verbose mode', e.g. they have
the flag --verbose passed to them.

This patch modifies all the underlying scripts in the build system to
inject --verbose into the command line when the environment variable
VERBOSE is set.

The environment variable VERBOSE is a CMake concept. CMake will
generate Makefile's that read this environment variable and try to
behave accordingly. Unfortunately, the generated ninja build systems
behave differently and will have to be invok [... cut]
commit 4971d2a0 (2018-01-11): cmake: Fix "make VERBOSE=1"

It is desired behaviour that when 'make VERBOSE=1' is invoked, the
underlying scripts will be invoked in 'verbose mode', e.g. they have
the flag --verbose passed to them.

This patch modifies all the underlying scripts in the build system to
inject --verbose into the command line when the environment variable
VERBOSE is set.

The environment variable VERBOSE is a CMake concept. CMake will
generate Makefile's that read this environment variable and try to
behave accordingly. Unfortunately, the generated ninja build systems
behave differently and will have to be invo [... cut]
closed 2018-01-11 by nashif