zephyrproject-rtos/zephyr #4864: cmake: hts221 sensor sample not working anymore

VVESTM, 2017-11-10:
Hi,

Since the latest updates (I am on sha1 994ec924e37e220add2bef8da3c90367704113af), I cannot run anymore the sample application for hts221 sensor on disco_l475_iot1 board. I have this error message in the console : 
   Could not get HTS221 device
I think this error is generic to all sensor sample applications.

Vincent

SebastianBoe, 2017-11-10:
A bug related to flashing that board was fixed 
in this commit
f0346bfac04704a8d9ca4bdcea718b66d11de01c
on master

Could you update master again and see if the issue is resolved now?

I am hoping this is a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/4844

erwango, 2017-11-10:
@SebastianBoe : I tested but unfortunately, this is another issue.

SebastianBoe, 2017-11-10:
I'll have a closer look then.

erwango, 2017-11-10:
Seems that generated .config file is fine.
Though, for some reason following files are not build:
  CC      drivers/sensor/hts221/hts221.o
  LD      drivers/sensor/hts221/built-in.o
  LD      drivers/sensor/built-in.o

SebastianBoe, 2017-11-10:
Good catch, that narrows it down!

SebastianBoe, 2017-11-10:
Right, got it, drivers/sensor/CMakeLists.txt is calling zephyr_sources instead of subdirectory.

Pushing a fix now.

Thanks a lot @erwango , I was debugging down the wrong path.

VVESTM, 2017-11-10:
Hi, 

Thanks a lot Sebastian !
I have tested with your patch and now, the hts221 sensor is working on this board.

It will be my model to port my PR related to vl53l0x sensor on cmake... (https://github.com/zephyrproject-rtos/zephyr/pull/829)

Regards
Vincent

SebastianBoe, 2017-11-10:
Thank you for reporting @VVESTM :)

nashif, 2017-11-10:
should be fixed

closed 2017-11-10 by nashif