zephyrproject-rtos/zephyr #4883: Should command line examples be "cut and paste" ready?

dbkinder, 2017-11-10:
With multi-line command line (bash) examples, it's convenient for users to just cut and paste them into a terminal window.  Our examples (mostly) start with a ``$`` on each line, making it inconvenient to cut and paste examples such as:
```
$ cd $ZEPHYR_BASE/tests/booting/stub

# Make a build directory, and use cmake to configure a Make-based build system:
$ mkdir build && cd build
$ cmake -DBOARD=arduino_101 ..

# Now run make on the generated build system:
$ make flash
``` 
vs.
```
cd $ZEPHYR_BASE/tests/booting/stub

# Make a build directory, and use cmake to configure a Make-based build system:
mkdir build && cd build
cmake -DBOARD=arduino_101 ..

# Now run make on the generated build system:
make flash
```

For examples that mix command and their output, I'm personally in favor of putting the ``$`` in, but for multi-line examples (without output) not using the ``$`` makes more sense.  

Your thoughts?

andrewboie, 2017-11-12:
> For examples that mix command and their output, I'm personally in favor of putting the $ in, but for multi-line examples (without output) not using the $ makes more sense.

+1

nashif, 2018-08-14:
Samples are mostly c/p ready, it depends on how complex the sample is, but for most part, you can just copy and paste them.

closed 2018-08-14 by nashif