zephyrproject-rtos/zephyr #4888: LwM2M: Fix BT device pending / retry errors

mike-scott, 2017-11-10:
In the bluetooth L2 layer the packets are being compressed prior to send.  This means we cannot merely resend the existing packets upon a packet timeout in the LwM2M / CoAP libraries.  Much like they do in the TCP subsystem, we need to make a clone of the original data for the pending / retry process.

As a result currently, we are seeing a double unref on NON-bluetooth devices:
https://github.com/zephyrproject-rtos/zephyr/pull/4391
And on bluetooth devices the retry process fails w/ ENOMEM (-12).

Vudentz, 2017-11-13:
Not following, is this suggesting to clone every single packet? Perhaps LwM2M should follow the TCP design and have its own poll to clone the packets for timeout handling.

Vudentz, 2017-11-13:
Btw, this is not unique to Bluetooth, afaik any driver using 6lo compression will change the packet.

galak, 2017-11-28:
Any updates on a fix for this?

mike-scott, 2017-11-28:
Let's mark this as 1.11.  I want to get the fix right for this.

carlescufi, 2018-01-23:
@mike-scott can we still expect a fix for this on 1.11?

mike-scott, 2018-01-24:
@carlescufi I think I'll defer this until post net-buf changes land.  This fix needs to go into the CoAP lib processing for retry so that it not only fixes LwM2M but also users of CoAP APIs as well.  Removing 1.11 target.

nashif, 2018-05-11:
what is the status of this one?

nashif, 2018-11-05:
@mike-scott ping

mike-scott, 2018-11-05:
@nashif This issue goes away in the move to sockets.  We can close.  Unless you want to track it separately.

closed 2018-11-05 by mike-scott