zephyrproject-rtos/zephyr #4902: Firmata support

askawu, 2017-11-13:
### Why
Firmata is a protocol for communicating with microcontrollers from software on a computer. There're many client implementations on the github. Most of them are different language binding, such as C++, Java, Python, Javascript, and Ruby. If Zephyr could support firmata as a server, existing client applications can communicate with Zephyr.

### How
- Evaluate the existing firmata server implementation. For example, https://github.com/firmata/arduino but it depends on some C++ standard library header files.
- Leverage the existing firmata server implementation or implement a new one from the ground up.

### Other
- Decouple the physical layer from firmata protocol. It should handle massage from different physical layers, such as serial port, ethernet, or usb.

pfalcon, 2018-09-06:
There were no comments to this over this time.

IMHO, Firmata support can come by means of some external 3rd-party library and doesn't have to be a part of Zephyr - after all, wide Firmata usage outside the Arduino community is questionable/limited.

@nashif : Suggested to close this, reopening later when usecases/stakeholders appear.

pfalcon, 2018-09-21:
To clarify, this was opened from Linaro side. But I see that internal ticket tracking this feature was closed. So, let me close this too to avoid ticket pile-up. This can be reopened if there's interest and good usecase to have this in tree (vs out of tree).

closed 2018-09-21 by pfalcon