zephyrproject-rtos/zephyr #4905: cmake: flashing for quark_se_devboard is broken

finikorg, 2017-11-13:
After cmake migration I cannot flash quark_se_c1000_devboard.

Error message is:
```
Open On-Chip Debugger 0.10.0-g7852ae77-dirty (2017-10-15-17:20)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
adapter speed: 1000 kHz
trst_only separate trst_push_pull
jtag_ntrst_delay: 300
Info : clock speed 1000 kHz
Info : JTAG tap: quark_se.cltap tap/device found: 0x0e765013 (mfg: 0x009 (Intel), part: 0xe765, ver: 0x0)
Enabling arc core tap
Info : JTAG tap: quark_se.arc-em enabled
Enabling quark core tap
Info : JTAG tap: quark_se.quark enabled
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0  quark_se.quark     quark_se   little quark_se.quark     running
 1* quark_se.arc-em    arc_quark  little quark_se.arc-em    halted
Info : JTAG tap: quark_se.cltap tap/device found: 0x0e765013 (mfg: 0x009 (Intel), part: 0xe765, ver: 0x0)
Enabling arc core tap
Info : JTAG tap: quark_se.arc-em enabled
Enabling quark core tap
Info : JTAG tap: quark_se.quark enabled
target halted due to debug-request at 0x0000fff0 in real mode
target halted due to debug-request at 0x0000fff0 in real mode
invalid subcommand "write_image erase /usr/local/src/viper/foss-rtos-rndis/samples/net/zperf/outdir/quark_se_c1000_devboard/zephyr/zephyr.bin 0x40030000"
in procedure 'flash'
Error: Command '['/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd', '-s', '/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts', '-f', 'quark_se_c1000_devboard/support/openocd.cfg', '-c', 'init', '-c', 'targets', '-c', 'targets 1', '-c', 'reset halt', '-c', 'flash write_image erase samples/net/zperf/outdir/quark_se_c1000_devboard/zephyr/zephyr.bin 0x40030000', '-c', 'reset halt', '-c', 'verify_image            samples/net/zperf/outdir/quark_se_c1000_devboard/zephyr/zephyr.bin 0x40030000', '-c', 'reset run', '-c', 'shutdown']' returned non-zero exit status 1
Re-run with KBUILD_VERBOSE=1 for a stack trace.
zephyr/cmake/flash/CMakeFiles/flash.dir/build.make:57: recipe for target 'zephyr/cmake/flash/CMakeFiles/flash' failed
make[4]: *** [zephyr/cmake/flash/CMakeFiles/flash] Error 1
CMakeFiles/Makefile2:2164: recipe for target 'zephyr/cmake/flash/CMakeFiles/flash.dir/all' failed
make[3]: *** [zephyr/cmake/flash/CMakeFiles/flash.dir/all] Error 2
CMakeFiles/Makefile2:2171: recipe for target 'zephyr/cmake/flash/CMakeFiles/flash.dir/rule' failed
make[2]: *** [zephyr/cmake/flash/CMakeFiles/flash.dir/rule] Error 2
Makefile:456: recipe for target 'flash' failed
make[1]: *** [flash] Error 2
/home/niko/bin/zmake.sh:13: recipe for target 'flash' failed
make: *** [flash] Error 2

```

closed 2017-11-13 by nashif

mnkp, 2017-11-13:
I see the same problem when flashing Atmel SAM E70 Xplained development board. It is reported however slightly differently
```
invalid command name "flash write_image erase /home/<user>/work/zephyr/samples/hello_world/build/zephyr/zephyr.bin 0x400000"
Error: Command '['/home/<user>/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/openocd', '-f', '/home/<user>/work/zephyr/boards/arm/sam_e70_xplained/support/openocd.cfg', '-c', 'init', '-c', 'targets', '-c', 'reset halt', '-c', 'flash\\ write_image\\ erase\\ /home/<user>/work/zephyr/samples/hello_world/build/zephyr/zephyr.bin\\ 0x400000', '-c', 'reset halt', '-c', 'verify_image\\ \\ \\ \\ \\ \\ \\ \\ \\ \\ /home/<user>/work/zephyr/samples/hello_world/build/zephyr/zephyr.bin\\ 0x400000', '-c', 'atsamv\\ gpnvm\\ set\\ 1', '-c', 'reset run', '-c', 'shutdown']' returned non-zero exit status 1
```
The difference are backslashes in 
```
'-c', 'flash\\ write_image\\ erase\\ /home/<user>/work/zephyr/samples/hello_world/build/zephyr/zephyr.bin\\ 0x400000'
```
Re-running the openocd command by hand with backslashes removed fixes the problem. Re-running the openocd command by hand keeping backslashes intact results in the same error as reported by make.

My system is Kubuntu 17.04 with all latest updates installed.

SebastianBoe, 2017-11-13:
@mnkp : Just tried locally on commit 539d2af654c4089cbdacdbe69676fdf0c95eb1f3.

I'm not seeing the backslash problem.

There was a bug fixed with the same symptoms as you are reporting in this commit: dfc13b21fced7504737dd396dd4900156cf21475

Could you post the steps to reproduce and what (clean) version of master you are using?

mnkp, 2017-11-13:
@SebastianBoe I pulled 'master' last Friday. With the current version the problem if fixed. I am able to flash my board successfully. Thanks for help!

I have an unrelated question: If you were able to flash SAM E70 Xplained board does Hello World application work for you? In my case the application crashes (an exception handler is invoked) long before the main() is reached. With the old make system all works fine.

SebastianBoe, 2017-11-13:
I don't have the HW unfortunately. But if this is a regression due to the CMake migration then I can debug it without the HW. Please report a bug when you are sufficiently convinced there is no user error and I will gladly debug :)

mnkp, 2017-11-14:
I had a closer look yesterday and found the issue. I've just posted the fix: #4948

PR #4908 (2017-11-13): cmake: Fixes quark_se_c1000_devboard flashing
Fixes: #4905
commit 3c6dee35 (2017-11-13): cmake: Fixes quark_se_c1000_devboard flashing

Fixes: #4905