zephyrproject-rtos/zephyr #4911: Filesystem support for qemu

jhedberg, 2017-11-13:
I'd like to be able to do something like "make run FS=/some/host/path" and have the directory structure under /some/host/path become available inside qemu as persistent storage that can be accessed using the Zephyr file system APIs.

andrewboie, 2017-11-28:
(01:14:40 PM) Anas: any idea if there is a way to emulate internal flash of an mcu with qemu? (https://github.com/zephyrproject-rtos/zephyr/issues/4911)
(01:18:02 PM) apboie: Anas: I know you can pass a disk image file to QEMU to use
(01:18:35 PM) apboie: -drive option
(01:18:59 PM) apboie: but that may or may not work, probably emulates a SATA device or something, not some memory-mapped flash
(01:21:01 PM) leandro.pereira: you can pass different interface types
(01:21:39 PM) leandro.pereira: -drive if=pflash,file=/path/to/some/flash/image.img
(01:22:08 PM) leandro.pereira: (not sure if pflash is what you're looking for, though.)
(01:25:18 PM) apboie: that seems the right ballpark

carlescufi, 2020-03-17:
N/A

closed 2020-03-17 by carlescufi

carlescufi, 2021-06-11:
@mglettig this is supported in [native_posix](https://docs.zephyrproject.org/latest/boards/posix/native_posix/doc/index.html#host-based-flash-access) instead, maybe that would be a solution for you?

yashi, 2021-06-12:
@mglettig, it's quite easy to create a `genlittlefs` tool using its library.  You can point to generated file system image when you run qemu instance.

mglettig, 2021-06-14:
> @mglettig this is supported in [native_posix](https://docs.zephyrproject.org/latest/boards/posix/native_posix/doc/index.html#host-based-flash-access) instead, maybe that would be a solution for you?

Good idea. Unfourunately this is not possible in my case due to this bug here: https://github.com/zephyrproject-rtos/zephyr/issues/34226

@yashi This sounds very promising. How can I point to the generated file system image from QEMU? And how will my Zephyr application find this filesystem image so I can mount the Little FS filesystem? Any hints would be welcome.

yashi, 2021-06-15:
@mglettig Oops, I meant native_posix instance, zephyr.exe.  Not qemu.  Sorry about that.