zephyrproject-rtos/zephyr #4912: Fix ZEPHYR_GCC_VARIANT

SebastianBoe, 2017-11-13:
Selecting toolchains and SDKs with environment variables is a broken incomprehensible mess.

Find and implement a design that is understandable and satisfies the requirements of today and the near future (multiple SDK providers).

GautierAtWork, 2017-11-13:
Using environment variables for dealing with different tools versions is a generic approach, which is used  also in large scale:
http://modules.sourceforge.net/
https://en.wikipedia.org/wiki/Environment_Modules_(software)

It is not uncommon to see hundreds of tools with tens of versions each managed this way...

SebastianBoe, 2017-11-13:
I'm sorry @GautierAtWork , I should have been more specific, let me rework the issue text.

SebastianBoe, 2017-11-13:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/4918

closed 2017-11-14 by SebastianBoe