zephyrproject-rtos/zephyr #4917: CMake migration regressed (removed) targets "initconfig", "outputexports"

pfalcon, 2017-11-13:
As was expected, CMake migration broke building of 3rd-party projects with Zephyr ports which call out to Zephyr build system from their own (cross-platform) build systems.

Here's example: https://github.com/micropython/micropython/blob/master/ports/zephyr/Makefile#L70
~~~
$(MAKE) -f Makefile.zephyr BOARD=$(BOARD) CONF_FILE=$(CONF_FILE) initconfig outputexports lib
~~~

How this is supposed to work: The basic requirements are: 1) being able to build 3rd-party app sources with the same compiler and compiler options as Zephyr uses; 2) being able to include zephyr headers in the app sources; 3) being able to pass an application library for the final Zephyr link stage. The process is thus:

1. Generate any Kconfig headers with config options, to make <zephyr.h>, etc. inclusion possible 
- initconfig target.
2. Generate a special file Makefile.exports which contains dump of all make vars used for Zephyr builds (CC=... , CFLAGS=..., etc.) - outputexports
3. Being able to pass ALL_LIBS var to Zephyr's Makefile (and LDFLAGS_zephyr) to Zephyr "all" target (https://github.com/micropython/micropython/blob/master/ports/zephyr/Makefile.zephyr#L24)

It's unclear how to reinstate this functionality with CMake, please suggest.

mbolivar, 2017-11-13:
@pfalcon Our out of tree build scripts were also (as expected) broken by this, and I'll be looking at that this week, so I hope it's OK if I jump on in hopes of sharing common ground / solutions.

> Generate any Kconfig headers with config options, to make <zephyr.h>, etc. inclusion possible
> initconfig target.

I assume you've taken a look at this information:
http://docs.zephyrproject.org/application/application.html#application-specific-code

It might help to explain why it's not satisfactory for your use case.

> Generate a special file Makefile.exports which contains dump of all make vars used for Zephyr builds (CC=... , CFLAGS=..., etc.) - outputexports
> Being able to pass ALL_LIBS var to Zephyr's Makefile (and LDFLAGS_zephyr) to Zephyr "all" target (https://github.com/micropython/micropython/blob/master/ports/zephyr/Makefile.zephyr#L24)

I wonder if having a Make-specific solution is really best now that the generated build system could be Ninja files, IDE solution files, etc.

One thing I was thinking of was extending the build system to use CMake cache entries to store additional variables (like FLASH_SCRIPT and its friends) for my scripts to access after the Zephyr build system has been generated.

https://cmake.org/cmake/help/v3.8/command/set.html

Do you think something like that could work for you, too?

pfalcon, 2017-11-13:
I'm looking at samples/application_development/external_lib/, but I don't see anything helpful so far. Besides CMake generated makefiles being not too readable, that sample offers opposite flow, where Zephyr's build system calls into 3rd-party build system. That's not flexible enough, what's needed is "inversion of control" solution, where 3rd-party build system calls into Zephyr's to run a particular stage of Zephyr's build process.

SebastianBoe, 2017-11-13:
> One thing I was thinking of was extending the build system to use CMake cache entries to store additional variables (like FLASH_SCRIPT and its friends) for my scripts to access after the Zephyr build system has been generated.

Changing certain variables into Cache variables to be able to extract them after invoking CMake seems like a sane approach to me.

@pfalcon : Thank you for trying to define requirements. The ability to integrate with third-party projects is important. I was under the impression that both mcuboot and micropython had the same structure as external_lib, I was obviously mistaken.

You are right in that the new build system does not support targets that generate build information as KBuild does. I don't see any reason why such targets couldn't be implemented, similair to "make usage", but instead of writing BOARD information to stdout, it writes machine-readable build information to a file. But I would like to have a closer look at the micropython use-case before implementing this.

I thought that either a third-party build system would not care about how Zephyr built it's code, and simply invoke the zephyr build system and use it's archive's / hex files. Or that the third-party build system would be invoked by Zephyr and import build information as external_lib does. The use-case
you described where the third-party build system is the toplevel build system and invokes Zephyr to retrieve build information instead of binaries is not covered.

I am going to work on the use-case of multi-application builds towards the next release. I'll make sure I understand your use-case and that whatever solution is found there either covers, or at least is not in the way of your requirements.

SebastianBoe, 2017-11-13:
> It's unclear how to reinstate this functionality with CMake, please suggest.

If one were to go down that road I might define a "custom_target" (see zephyr/cmake/usage/CMakeLists.txt). And have that custom_target call the zephyr_get_* API and write that information to a file.

pfalcon, 2017-11-13:
@mbolivar : 

Sure, I appreciate the comments and ideas!

> I assume you've taken a look at this information: http://docs.zephyrproject.org/application/application.html#application-specific-code

I now did. Yes, I don't see immediately how to make use of that due to issue mentioned in my previous comment (which I posted before seeing yours): they seem to work for usecase when 3rd-party build is integrated into Zephyr's CMake build, micropython, jerryscript need it vice-versa.

As far as I can tell, a solution might be to include these CMake functions (zephyr_get_compile_options_as_string(), etc.) by default into each CMakefile, under a specific Make target name, with that target to output their values e.g. to a file.

> I wonder if having a Make-specific solution is really best now that the generated build system could be Ninja files, IDE solution files, etc.

My answer is "yes" ;-).

> One thing I was thinking of was extending the build system to use CMake cache entries to store additional variables (like FLASH_SCRIPT and its friends) for my scripts to access after the Zephyr [...]

Looking at CMakeCache.txt, its format is not make (or shell) friendly. Of course, it can be preprocessed on 3rd-party app side. That all sounds a bit hacky though, I'd hope we could make it better (actually, maybe even improve how it was before). Well, I know nothing about CMake (and find it to be rather non-self-descriptive, looking and both input and output files), so can only guess what's possible to do and what isn't.

pfalcon, 2017-11-13:
@SebastianBoe :

> If one were to go down that road I might define a "custom_target" (see zephyr/cmake/usage/CMakeLists.txt). And have that custom_target call the zephyr_get_* API and write that information to a file.

Trying quick hack on that:

~~~
include($ENV{ZEPHYR_BASE}/cmake/extensions.cmake)

message("outputexports goes here")

zephyr_get_compile_options_as_string(cflags)

message("CFLAGS=${cflags}")
~~~

Leads to:

~~~
outputexports goes here
CMake Error at /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/cmake/extensions.cmake:158 (get_property):
  get_property could not find TARGET zephyr_interface.  Perhaps it has not
  yet been created.
Call Stack (most recent call first):
  /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/cmake/extensions.cmake:183 (zephyr_get_compile_options)
  /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/cmake/usage/outputexports.cmake:7 (zephyr_get_compile_options_as_string)
~~~

And indeed, I see references to zephyr_interface only in extensions.cmake, and nothing seems to "define" it there. (Per my understanding, there should be "add_<something>" call to define a target, a-la https://cmake.org/cmake/help/latest/manual/cmake-buildsystem.7.html#binary-targets)

mbolivar, 2017-11-13:
> I was under the impression that both mcuboot and micropython had the same structure as external_lib, I was obviously mistaken

By the way, mcuboot is just a "normal" Zephyr application, so the external_lib extras are not needed. Simply a single application CMakeLists.txt will suffice. I'm working on that this week.

mbolivar, 2017-11-13:
> Looking at CMakeCache.txt, its format is not make (or shell) friendly. Of course, it can be preprocessed on 3rd-party app side. That all sounds a bit hacky though, I'd hope we could make it better [...]

Well, it seems you're off on another direction given the above, but `cmake -L` can be used to print the values of cache variables, rather than parsing CMakeCache.txt by hand:

https://cmake.org/cmake/help/v3.8/manual/cmake.1.html

(FWIW, I think the ability to call into the Zephyr build might be interesting for us as well, but I'd say it's a lower priority at the moment for getting mcuboot-based apps working cleanly in Zephyr than teaching the flashing scripts about device tree, which I'm hoping to find time to work on this week as well.)

nashif, 2017-11-13:
Can someone please write up the requirement in a clear way without making assumptions and without asking for some "features" we had in kbuild that were pretty much a work-around. We want to do this right this time and it should not be a forward ported hack. Please.

mbolivar, 2017-11-14:
> Can someone please write up the requirement in a clear way without making assumptions and without asking for some "features" we had in kbuild that were pretty much a work-around. We want to do this right this time and it should not be a forward ported hack. Please.

The main thing I've still not got a replacement that Makefile.exports previously enabled is getting the board-specific information provided to Zephyr's build system to flash the board.

This let me write some build and flash scripts that tie together an mcuboot build, a zephyr application build, and use information available in device tree to flash them at the right offsets, doing a full chip erase before flashing mcuboot.

These scripts either operate standalone, or generate a shell script in a build directory containing the mcuboot and application build artifacts which does the same thing (so the binaries can be distributed, used in build testing, etc.)

This seems to be different than what Paul needs.

I think my plan for now to replicate this feature is:

1. teach the (upstream, Python) runner scripts about device tree, so the generated build system can flash binaries at the right offsets
2. teach the Python scripts to generate shell scripts in a similar way our out of tree ones do (I think this is likely to be convenient for other people who want to distribute a binary and a way to flash it without shipping the entire zephyr build system)
3. teach the Python scripts to take arguments which do or do not do a chip erase, and configure the builds accordingly
4. stitch it all together in my own application builds using only upstream mechanisms

If anybody has any comments about these requirements' suitability for upstream inclusion or my plan of implementation in 1, 2, 3, that'd be welcome.

pfalcon, 2017-11-14:
@nashif :

> Can someone please write up the requirement in a clear way without making assumptions and without asking for some "features" we had in kbuild that were pretty much a work-around. We want to do this right this time and it should not be a forward ported hack. Please.

That's exactly what I didn in the top description of the ticket, namely:

> The basic requirements are: 1) being able to build 3rd-party app sources with the same compiler and compiler options as Zephyr uses; 2) being able to include zephyr headers in the app sources; 3) being able to pass an application library for the final Zephyr link stage.

I definitely agree that it doesn't have to use the same make targets/variables like before. Maybe, it doesn't even have to be a "make" target/variable (maybe CMake ones).

pfalcon, 2017-11-14:
> The main thing I've still not got a replacement for that Makefile.exports previously enabled is getting the board-specific information provided to Zephyr's build system to flash the board.

I don't even think that far. My initial goal would be to get a binary build again. @mbolivar , I surely appreciate your work on standardizing flashing support in Zephyr, and surely agree that it would be to "pass thru" it to 3rd-party apps, with additional params like partition layout. All in all, that seems an additional, orthogonal requirement to my list above.

pfalcon, 2017-11-14:
@SebastianBoe :

> I thought that either a third-party build system would not care about how Zephyr built it's code, and simply invoke the zephyr build system and use it's archive's / hex files. Or that the [...]
you described where the third-party build system is the toplevel build system and invokes Zephyr to retrieve build information instead of binaries is not covered.

To discuss this further:

Method 1: "a third-party build system would not care about how Zephyr built it's code" - This is not realistic, because almost any 3rd-party project would need to build its sources too, and of course, would need to build them into compatible to Zephyr object files. So, a 3rd-party app would need to known (complete) CC, CFLAGS to build its sources.

Method 2: "Or that the third-party build system would be invoked by Zephyr and import build information as external_lib does." - This is not practical for cross-platform projects with sufficiently complex own build systems. Besides pure complexity of integrating a build system with own custom code preprocessors/generators, etc., there's also a organizational issue that it would make a Zephyr port of the project very different from other ports, up to a way that project maintainers being not able to understand/support it. Both MicroPython and JerryScript projects are such - portable cross-platform projects, with their own non-trivial build systems. (And indirectly affects Zephyr.js, which is while Zephyr-specific, uses JerryScript as a component).

SebastianBoe, 2017-11-14:
> So, a 3rd-party app would need to known (complete) CC, CFLAGS to build its sources.

It doesn't always need to know this does it? In the use-case where the 3rd-party app already
 knows how to build for an SoC it will already have figured out what CC and CFLAGS to use
wouldn't it? I thought that two archives could be linked together without using the same 
CC and FLAGS? Perhaps there exist some small set of flags that the third-party actually
needs to know, like -mabi, that Zephyr could export.

> 1. Generate any Kconfig headers with config options, to make <zephyr.h>, etc. inclusion possible - initconfig target.

I think this particular requirement is already supported. After running CMake on an application it will
have generated Kconfig headers even before Make is invoked.

pfalcon, 2017-11-14:
> > Generate any Kconfig headers with config options, to make <zephyr.h>, etc. inclusion possible - initconfig target.

> I think this particular requirement is already supported. After running CMake on an application it will
have generated Kconfig headers even before Make is invoked.

Ack, thanks!

pfalcon, 2017-11-14:
>> So, a 3rd-party app would need to known (complete) CC, CFLAGS to build its sources.

> It doesn't always need to know this does it? In the use-case where the 3rd-party app already
knows how to build for an SoC it will already have figured out what CC and CFLAGS to use
wouldn't it?

But how a 3rd-party app may know that? Zephyr is a portable RTOS, so a Zephyr application should be able to build for any target supported by Zephyr (including future).

I didn't mention that, but all the requirements and points above come from enough (year+) experience. A case you talk about reminds me about how the initial JerryScript port was done: 1) It supported just a couple of boards. 2) It had exactly own hardcodes for CC/CFLAGS for a particular board. One of them was frdm_k64f. As soon as its support in Z was optimized (to use Cortex-M4F compile options), the "integration" broke, because JrS side used generic Cortex-M3 options (leading to incompatible calling conventions).

Summing up, a portable Zephyr app can't "know" what CC/CFLAGS options to use, the only reliable way is to ask Zephyr.

nashif, 2017-11-14:
> It doesn't always need to know this does it? In the use-case where the 3rd-party app already
> knows how to build for an SoC it will already have figured out what CC and CFLAGS to use
> wouldn't it?

The support for a specific SoC will come from Zephyr, not from the middleware you are trying to build, the middleware might have notion of what hardware it will be running on, so this needs to come from Zephyr.

mbolivar, 2017-11-14:
> I don't even think that far. My initial goal would be to get a binary build again. @mbolivar , I surely appreciate your work on standardizing flashing support in Zephyr, and surely agree that it [...]

@pfalcon Thanks and sure, which is why I wrote "This seems to be different than what Paul needs". I only joined this issue as the top-level description appeared related. I now see that it isn't, so I'll leave this thread.

pfalcon, 2017-11-17:
@SebastianBoe :

Any further suggestions re: (https://github.com/zephyrproject-rtos/zephyr/issues/4917#issuecomment-343999824)

>Trying quick hack on that:

~~~
include($ENV{ZEPHYR_BASE}/cmake/extensions.cmake)
message("outputexports goes here")
zephyr_get_compile_options_as_string(cflags)
message("CFLAGS=${cflags}")
~~~

> Leads to: [error]

(or suggestions for any other direction)

SebastianBoe, 2017-11-17:
It's not clear how that code snippet has been integrated. But I am going to assume that it is acting
as the toplevel/application CMakeLists.txt file, if so it needs to be placed after the boilerplate that is in every application. If it is not structured like that then that would explain the error.

I would do something like this if I were you:

Develop an application CMakeLists.txt file.
To satisfy this requirement: "being able to pass an application library for the final Zephyr link stage."
Either use a cmake INTERFACE library, like we do in external_lib. Or use this less
elegant, but simpler technique: https://github.com/zephyrproject-rtos/zephyr/blob/master/ext/hal/qmsi/CMakeLists.txt#L12

The third-party build system would then do something like:

call "cmake" on a custom-designed application CMakeLists.txt file to generate zephyr header files (And maybe also a Makefile.export-like file)
Possibly call "make Makefile_export" to generate a Makefile.export-like file.
Have your own build system build your application library.
Call "make" now with no targets, it should now find your application library and link it in at the end of the build.

pfalcon, 2017-11-17:
> It's not clear how that code snippet has been integrated.

In the least-resistance way following your pointer to cmake/usage/ as implementing something similar:

1. The quoted code content is in cmake/usage/outputexports.cmake
2. cmake/usage/CMakeLists.txt patched with:

~~~
+add_custom_target(
+  outputexports
+  ${CMAKE_COMMAND} -P ${CMAKE_CURRENT_SOURCE_DIR}/outputexports.cmake
+  )
~~~

pfalcon, 2017-11-17:
> Develop an application CMakeLists.txt file.

I'll try to follow your suggestion, but my aim is not unexpectedly to just get back the behavior which was available previously: running "make outputexports" for any Zephyr app produces the needed list, without any involvement from app's side.

SebastianBoe, 2017-11-17:
> cmake/usage/CMakeLists.txt patched

Ah, right, so turns out "usage" is not so similair after all. Invoking CMake again as usage does gives us a new CMake build system that doesn't know anything.

We need another mechanism, perhaps using ExternalProject with a dummy build system that exists only to receive build information from the Zephyr build system and write it to some file for consumption by the third-party build system.

Or maybe custom_target with a python script is even easier.

pfalcon, 2017-11-17:
Ok, I was able to make MicroPython port build against CMake, the patch is here https://github.com/pfalcon/micropython/commit/c021057fbe49dbdd46ccd2506f9ce2034bf5e257 . As can be seen, it effectively follows application_development/external_lib very closely. I'll refrain from comments so far, will need to play with it more and think about, then update JerryScript too, to gather my thoughts on it.

pfalcon, 2017-11-23:
Also posted an initial version of the JerryScript port update: https://github.com/jerryscript-project/jerryscript/pull/2120

pfalcon, 2017-12-01:
Ok, so the current status of updating 3rd-party projects for Zephyr CMake build system:

* MicroPython changes are merged in https://github.com/micropython/micropython/commit/a036554a77c50fc4aabf056d93f7b3170ee4af21 (with a recent fix up in https://github.com/micropython/micropython/commit/bb047558da6d7e1857e1d4c79547776976353cd3)
* Patch for JerryScript submitted: https://github.com/jerryscript-project/jerryscript/pull/2120
* Zephyr.js update is handled by its team: https://github.com/01org/zephyr.js/pull/1665 . @jimmy-huang, if you have any comments to share, that would be useful.

pfalcon, 2017-12-01:
So, it's fair to say that the original 3 requirements are possible with new CMake-based build system, and I ticked checkboxes in the top description.

But: doing the "being able to build 3rd-party app sources with the same compiler and compiler options as Zephyr uses" requirement takes adding an adhoc custom target to project's (each 3rd-party project!) CMakeLists.txt. Fairly speaking, each project would need to add a CMakeLists.txt anyway, and (general) magic to get Zephyr's CC/CFLAGS can be looked up in samples/application_development/external_lib.

However, it still requires some time and CMake familiarity to put that all together, and every project might end up doing that somewhat inconsistently (like was done for MicroPython vs JerryScript), which also complicates problem understanding/knowledge sharing.

So, I'd still appreciate if an approach to make a generic "outputexports" targets available for any project was considered, and added 4th requirement:

"being able to do all of the above with the "default" CMake setup, without adding custom targets to every 3rd-party project CMakeLists.txt."

But otherwise, setting this a low priority, hoping that @SebastianBoe might look into this next couple of months, in the attempt to do it "right this time", as suggested by @nashif .

jimmy-huang, 2017-12-02:
@pfalcon I have ZJS building with cmake and I used the samples/application_development/external_lib/ as a reference, but I am using ExternalProject_Add() instead of custom_target but so far it's working, I am  but I am not sure if is the ideal way, because we want to build jerryscript as as two separate static libaries (w/without the parser), and pick which one to link when we build that app.  So far I have not find a good way to force a rebuild of that particular custom library, i basically have to wipe the whole builddir to rebuild everything.  So feel free to take a look at my patch.

pfalcon, 2019-05-28:
This is serious issue, potentially affecting any existing project ported to Zephyr, e.g. https://github.com/zephyrproject-rtos/zephyr/pull/16427

This is not "enhancement", this is regression introduced during cmake migration.

PR #22020 (2020-01-23): cmake: Add option for exporting build metadata to Make
Add an opt-in feature that will generate a Makefile with build
variables like CC, and KBUILD_CFLAGS for consumption by third-party
Make-based build systems.

This emulates the 'outputexports' target that KBuild supported and is
supported for the same reasons that KBuild supported it. Easier
integration with third-party build systems.

This fixes #4917
closed 2020-01-23 by nashif