zephyrproject-rtos/zephyr #4925: gen_inc_file test pollutes source directory

andrewboie, 2017-11-13:
Output of 'git status' after running sanitycheck:

```
$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
Untracked files:
  (use "git add <file>..." to include in what will be committed)

	tests/application_development/gen_inc_file/src/file.bin.gz.inc
	tests/application_development/gen_inc_file/src/file.bin.inc
```

The build system needs to place all generated artifacts in the output directory, not mixed in with the source tree

nashif, 2017-11-13:
wasn't this just fixed by 965f78088fd57bfabfb74943b22613973d69f85b?

andrewboie, 2017-11-13:
might be. let me rebase my tree and re-test

andrewboie, 2017-11-13:
@nashif no this is not fixed

reproduce by running "sanitycheck -i"

andrewboie, 2017-11-13:
This is what I am seeing now:

```
	samples/application_development/external_lib/mylib/src/mylib_project-stamp/
	samples/application_development/external_lib/mylib/tmp/
```

SebastianBoe, 2017-11-14:
Sorry about creating paperwork, should of course not have been like this in the first place.

PR #4950 (2017-11-15): samples: external_lib: Build binaries in the build directory
The "external" Makefile in the sample "external_lib" did not support out-of-source builds. This meant hat the working directory would get polluted with source files when the sample was built. This commit rewrites the Makefile to support out-of-source builds and updates the sample's CMakeLists.txt to
specify where the build directory should be.

The working directory is no longer polluted with untracked files when the sample is built.

This fixes #4925
closed 2017-11-15 by nashif