zephyrproject-rtos/zephyr #4927: Hard Fault when no device driver is setup for ENTROPY results

dleach02, 2017-11-13:
A bug was introduced into the build system (CMAKE) for KW41Z that caused the build to not have a default device driver for entropy. The initial problem is that there doesn't seem to be a way to catch a build problem here since the binding of a driver is done at run time. In sys_rand32_get(), there is a call to get the device binding to the CONFIG_ENTROPY_NAME. If the build system has an error (or an error has been introduced), this call to device_get_binding() will return a NULL because there is no matching name in the device list and there is no error check in this code to ensure a proper value was returned. Eventually a hard fault will occur because the code will later want to call through the device entry to make a call to the support function but this is random address.

Ideally, we would have a compile/link time error when a required device is not included in the system but I'm not sure how to do that given how devices are added to the init section to be invoked/bind at runtime. Having an assert/error message on a null value returned from device_get_binding() is the next best thing although the CI build system won't catch introduced errors as this will only show up during code execution.

SebastianBoe, 2017-11-14:
This is really bad.

Did Kbuild do this better, or has the migration just initially introduced a lot of bugs of this nature?

Any ideas for solutions?

I'm not familiar with the device binding mechanism.

Is it feasible to offer an implementation of device_get_binding() that is evaluated and checked at build time instead of run time?

SebastianBoe, 2017-11-14:
@dleach02 : Could you rename the title to make this not specific to entropy?

lpereira, 2017-11-14:
I'll look into this.  In any case, an assertion in `sys_rand32_get()` is now in place, so at least it'll fail in runtime with a message to check the configuration.

dleach02, 2017-11-15:
I'm not sure there is a solution for this to generically deal with the problem of not defining a binding. I created the issue to see if someone has any idea how we could catch this via the build system instead of runtime crashes.

lpereira, 2017-11-15:
I think I found the bug in the `Kconfig` files:

KW41Z4 has a hardware number generator that is supported by Zephyr (it's based off of the MCUX HAL, using the TRNG variant), so it should provide an entropy driver.  In `arch/arm/soc/nxp_kinetis/kwx/Kconfig.soc`, one can se that, if `CONFIG_SOC_MKW41Z4` is selected, `CONFIG_HAS_MCUX_TRNG` is also selected.

The driver is only built if `CONFIG_ENTROPY_GENERATOR` is set, however.  It's not by default. 

However, `CONFIG_ENTROPY_DEVICE_RANDOM_GENERATOR` is selected by default in the "Random generator" choice (`subsys/random/Kconfig`); however, `CONFIG_ENTROPY_DEVICE_RANDOM_GENERATOR` depends on `CONFIG_ENTROPY_HAS_DRIVER`, which is only set if `CONFIG_ENTROPY_GENERATOR` and the platform driver are selected.

I'm not sure now what happens if a choice is the default but it has a dependency that's not selected.

galak, 2017-11-28:
We added an assert in subsys/random/rand32_entropy_device.c for this, but should this be in device_get_binding()?  [ there might be some valid cases for getting null back ]

galak, 2017-11-28:
Added reference to #4928 for tracking purposes.

pfalcon, 2018-01-25:
> Hard Fault when no device driver is setup for ENTROPY results

From the description of it, it looks very similar to https://github.com/zephyrproject-rtos/zephyr/issues/3546 - there also, if no net iface is present, and network functions are called, it's easy to get crashes/lock-ups.

carlescufi, 2018-02-06:
@lpereira @dleach02 @nashif is this still relevant? I don't have the hardware to test if it still is.

dleach02, 2018-02-07:
I don't know. I entered the bug because it seemed like something that should be captured at build time instead of a hard crash at run time. Was there any specific changes made to allow this to occur?

SebastianBoe, 2018-02-08:
" it seemed like something that should be captured at build time instead of a hard crash at run time"
I agree. AFAICT this has been like this for a long time and still behaves like this. So we should keep this issue to track it.

nashif, 2018-11-21:
closing this issue, no activity for a while.

closed 2018-11-21 by nashif