zephyrproject-rtos/zephyr #4937: ESP32 can't boot

WeldonWangwang, 2017-11-14:
Hi,  

I followed the document at  [here](http://docs.zephyrproject.org/getting_started/getting_started.html) with ESP32, I tried the demo of `hello_world`, but when I run `make flash` after `make`, the terminal showed:

```
[  2%] Built target offsets
[  4%] Built target offsets_h
[ 37%] Built target kernel
[ 55%] Built target app
[ 56%] Built target linker_pass2_script
[ 57%] Built target linker_script
[ 89%] Built target zephyr
[ 92%] Built target zephyr_prebuilt
[ 98%] Built target kernel_elf
Scanning dependencies of target flash
[100%] Flashing esp32
Converting ELF to BIN
esptool.py v2.1
Flashing ESP32 on /dev/ttyUSB0 (921600bps)
esptool.py v2.1
Connecting....
Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Wrote 16384 bytes at 0x00001000 in 0.2 seconds (531.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
[100%] Built target flash
```

I found three questions:  

1. The `zephyr.bin` generated by make is 13,776 bytes, but the log shows it flashing 16384 bytes. I can't understand.  

2. The bin was flashed at 0x00001000, but 0x00001000 is the 2nd boot address, so what's the bin it flashed and how to flash the app_bin?  

3. I watched the log of UART after flash, but it showed the chip only had 1st boot, and stop work after 1st boot. The whole log is listed here:  

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffb0000,len:60
ho 0 tail 12 room 4
load:0x3ffb0040,len:640
load:0x3ffb02c0,len:1600
ho 0 tail 12 room 4
load:0x40080000,len:1024
load:0x40080400,len:60
load:0x4008043c,len:10268
ho 0 tail 12 room 4
load:0x400c0000,len:0
entry 0x400805f0
```

I want to know if it's because my wrong setting or operation.    

Thanks!!!

SebastianBoe, 2017-11-14:
> The zephyr.bin generated by make is 13,776 bytes, but the log shows it flashing 16384 bytes. I can't understand.

Not sure how you are getting 13,776 bytes. I'm getting 

> /home/sebo/zephyr-sdk-0.9.2-rc5/sysroots/x86_64-pokysdk-linux/usr/bin/xtensa-zephyr-elf/xtensa-zephyr-elf-size b/zephyr/zephyr.elf 
>    text	   data	    bss	    dec	    hex	filename
>    9818	    756	   5760	  16334	   3fce	b/zephyr/zephyr.elf
> 

16334 bytes, which maybe becomes 16384 when rounded up to the page size?

SebastianBoe, 2017-11-14:
This PR should fix at least 1 bug that you are affected by. I don't have the HW so please retest with that patch to see if there are more bugs that are exposed by your test.

https://github.com/zephyrproject-rtos/zephyr/pull/4954

SebastianBoe, 2017-11-14:
Going to assume https://github.com/zephyrproject-rtos/zephyr/pull/4954 fixes this issue. Please re-open if the issue persists.

closed 2017-11-14 by SebastianBoe