zephyrproject-rtos/zephyr #4957: Add build targets for each explicit debug/flash option

galak, 2017-11-14:
Its possible that boards support multiple flash/debug choices (ie boards/arm/frdm_kw41z/board.cmake).  We need a way to allow for the user to do something like 'make debug-jlink' or 'make debug-daplink' and have the right thing happen if the board supports both.

SebastianBoe, 2017-11-14:
To implement this I suspect one would add a nested foreach loop in cmake/flash/CMakeLists.txt.

galak, 2017-11-14:
Was also thinking we'd add something like variables in the board.cmake files like BOARD_SUPPORTS_DEBUG_JLINK, BOARD_SUPPORTS_DEBUG_DAPLINK, etc that the board would define and maybe some means of saying which one was default so that 'make flash' would get the default choice.

nashif, 2017-11-14:
> Was also thinking we'd add something like variables in the board.cmake files like BOARD_SUPPORTS_DEBUG_JLINK, BOARD_SUPPORTS_DEBUG_DAPLINK, etc that the board would define and maybe some means of saying which one was default so that 'make flash' would get the default choice.

maybe add those to the board yaml file instead

galak, 2017-11-14:
> Was also thinking we'd add something like variables in the board.cmake files like BOARD_SUPPORTS_DEBUG_JLINK, BOARD_SUPPORTS_DEBUG_DAPLINK, etc that the board would define and maybe some means of saying which one was default so that 'make flash' would get the default choice.
> maybe add those to the board yaml file instead
>  
>   
>             
>  
> Write

do we p

> maybe add those to the board yaml file instead

Do we parse the board yaml in cmake?

SebastianBoe, 2017-11-14:
Not yet. So what @galak  proposed is easier to implement today.

galak, 2017-11-14:
Was also thinking that we should have the zephyr_flash_debug.py report back what flashers, debugers, etc are supported to give a list back to cmake.  So that way we don't need to update multiple places when adding a new debug/flash target.  I asked @mbolivar to try and look at adding that support to zephyr_flash_debug.py

mbolivar, 2018-06-15:
This is possible now with `west flash -r jlink`, `west flash -r <some-other-runner>`, etc.

closed 2018-06-15 by mbolivar