zephyrproject-rtos/zephyr #4959: Failure to install on Mac OS High Sierra (return of the same error)

salerio-gs, 2017-11-14:
Followed MacOS Setup as documented in the v1.9.1 Getting Started Guide

/usr/local/bin/ct-ng: line 7: MAKEFLAGS: command not found
/usr/local/bin/ct-ng: line 10: .NOTPARALLEL:: command not found
/usr/local/bin/ct-ng: line 13: MAKEFILE_LIST: command not found
/usr/local/bin/ct-ng: line 13: lastword: command not found
/usr/local/bin/ct-ng: line 13: abspath: command not found
/usr/local/bin/ct-ng: line 13: export: `CT_NG:=': not a valid identifier
/usr/local/bin/ct-ng: line 15: shell: command not found
/usr/local/bin/ct-ng: line 15: export: `CT_TOP_DIR:=': not a valid identifier
/usr/local/bin/ct-ng: line 19: export: `CT_LIB_DIR:=/usr/local/Cellar/crosstool-ng/1.23.0/lib/crosstool-ng-1.23.0': not a valid identifier
/usr/local/bin/ct-ng: line 20: export: `CT_DOC_DIR:=/usr/local/Cellar/crosstool-ng/1.23.0/share/doc/crosstool-ng/crosstool-ng-1.23.0': not a valid identifier
/usr/local/bin/ct-ng: line 23: export: `CT_VERSION:=crosstool-ng-1.23.0': not a valid identifier
/usr/local/bin/ct-ng: line 26: CT_LIB_DIR: command not found
/usr/local/bin/ct-ng: line 26: include: command not found

andrewboie, 2017-11-14:
seems like ct-ng is not being parsed as a Makefile and it thinks its a shell script instead.
Does "make -f /usr/local/bin/ct-ng" work?
Also did you build this ct-ng yourself or get it from brew? In my experience the brew build was horribly broken and it didn't work until I built it myself.

salerio-gs, 2017-11-14:
Yes make -f /usr/local/bin/ct-ng menu config works just fine. I think your deduction is correct, it does not think its a makefile. I am using cross tool-ng 1.23 from brew which the documentation says is the minimum required.

salerio-gs, 2017-11-14:
This command however does not, will try building crosstool-ng

Peters-MBP:CrossToolNG ps$ make -f /usr/local/bin/ct-ng build
/bin/bash: /usr/local/Cellar/crosstool-ng/1.23.0/lib/crosstool-ng-1.23.0/scripts/crosstool-NG.sh: Permission denied

salerio-gs, 2017-11-14:
Built my own crosstool-ng 1.23 from source, seems to work just fine, perhaps documentation should warn more strongly against using the homebrew build?

andrewboie, 2017-11-14:
> perhaps documentation should warn more strongly against using the homebrew build?

I concur. I believe at the time it was written, that Brew build wasn't yet available so issues with it were not known about.
Might be worth reporting upstream to the Brew team, it seems very very broken in ways that are trivial to reproduce

andrian7, 2017-11-24:
Quick fix for Permission denied:
`chmod +x /usr/local/Cellar/crosstool-ng/1.23.0/lib/crosstool-ng-1.23.0/scripts/crosstool-NG.sh`

tntclaus, 2018-04-02:
@andrian7 you made my day, man!

nashif, 2018-04-30:
duplicated by #6912

closed 2018-04-30 by nashif