zephyrproject-rtos/zephyr #4973: Provide Linux-style ERR_PTR/PTR_ERR/IS_ERR macros

jhedberg, 2017-11-14:
Linux has macros which enable functions returning pointers to indicate specific error codes in their return value rather than just returning NULL. It'd be nice to have similar functionality for Zephyr.

andrewboie, 2017-11-14:
It might be hard to do this since we do not virtualize memory, the Linux implementation assumes the range ((uint32_t)-MAX_ERRNO)-0xFFFFFFFF is never a valid memory address it seems.

There might be other approaches, such as stating that any returned pointer that isn't aligned to 4 bytes is a flag indicating that the real error value should be fetched from some other variable (thinking out loud here)

lpereira, 2017-11-15:
Can errno be used instead?  This is an idiom that I've seen used in the wild: `return errno = ESOMETHING, NULL;`.

jhedberg, 2017-11-15:
>  Can errno be used instead? This is an idiom that I've seen used in the wild: return errno = ESOMETHING, NULL;.

In some cases it could be an option. However it's not multi-thread safe, so it can't always be used. In general I'm not sure we should encourage errno usage because of its vulnerability to multi-threaded (or thread + ISR) race conditions.

andrewboie, 2017-11-15:
I'm not a fan of errno either unless we can migrate it to thread-specific storage.
Memory protection makes dealing with it a lot more complicated.

lpereira, 2017-11-16:
Yeah. That makes sense -- also, we don't seem to use errno that much in Zephyr anyway; we mostly use the error codes.

The idea to use unaligned pointers to signal errors has potential.  There's no need to have another errno-like variable to store the error, as it can be encoded in the pointer itself if it's not aligned. With a 4-byte alignment, only 2 bits dictate the alignment, so we have 30 bits for error codes;  that's more than plenty enough to encode errno codes.

In a way, this isn't much different than a function returning a negative errno on failure and 0 on success (sign bit being set to signal error, etc), with the exception that masking bits is more expensive than just checking for sign.

Just have to be careful to not use this for pointer types that do not have 4-byte alignments (`char *`, `short *`, etc).  In general, for returning a struct, this should be fine (unless it's packed and its `sizeof()` is smaller than `4`); maybe there's a way to `static_assert()` that these macros will only be used with appropriately-sized types?

jhedberg, 2020-02-11:
I'll close this, since after the discussion in #22684 we decided not to pursue these Linux-style APIs.

closed 2020-02-11 by jhedberg