zephyrproject-rtos/zephyr #4975: Getting started documentation for Mac OS X inconsistent

salerio-gs, 2017-11-14:
The following is from the above for v1.9.1. It states that there is a ct-ng configuration for ARM toolchain generation, there is no such configuration.

> In the Zephyr kernel source tree we provide two configurations for both ARM and X86 that can be used to pre-select the options needed for building the toolchain. The configuration files can be found in $ZEPHYR_BASE/scripts/cross_compiler/.
> 
> Currently the following configurations are provided:
> 
> i586.config: for standard ABI, for example for Galileo and qemu_x86
> iamcu.config: for IAMCU ABI, for example for the Arduino 101
> nios2.config: for Nios II boards

nashif, 2017-11-14:
For ARM we recommend using https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads, but this needs to be fixed, agree.

salerio-gs, 2017-11-14:
Thx, will take that advice ( new to Zephyr )

PR #4976 (2017-11-15): doc: fix reference to CTNG supported toolchains
Fixes #4975
closed 2017-11-15 by nashif