zephyrproject-rtos/zephyr #4989: Bluetooth: PTS fails to connect to the Zephyr

MariuszSkamra, 2017-11-15:
HW
- Tester application running on QEMU
- Bluetooth USB dongle or Zephyr based (hci_uart) bt dongle
or 
- Tester combined build on nrf52 board

PTS fails to connect to the Zephyr. Note that only few test cases are failing due to this, the rest of tests is passing. PTS orders IUT to enable advertising, then tries to connect. Connection fails to be established. 

Steps to reproduce:
- Run one of the failing test cases
- PTS will try to connect to the IUT
- Connection will fail (Disconnection with Reason: Connection Failed to be Established)

Regression found in following test cases:
GATT/SR/GAW/BI-01-C
GATT/SR/GAS/BV-01-C

Git bisect testing pointed **a74feaaf557012d42ace232d0411d9275e21e066** as first bad commit. Tests are passing on build before this commit.

I enclose debug logs, btmon and Protocol Viewer logs running on current master - new (7f6070fb6c3b5075040c3de7880edaad697de762) and old, before the change (on 7762578de4b7bc5784b08dc0a71a64615d82a471).

kolodgrz, 2017-11-15:
This also affects GATT/SR/GAW/BV-02-C, GATT/SR/GAD/BV-01-C

MariuszSkamra, 2017-11-21:
I tested this with privacy disabled (CONFIG_BT_PRIVACY=n) and it passed

MariuszSkamra, 2017-11-21:
So there is a connection issue only if privacy is enabled

MariuszSkamra, 2017-11-23:
Steps to reproduce:
1. Start samples/bluetooth/peripheral application in QEMU
2. Try to connect

Expected result:
Connection successful

Actual result:
Connection failed to be established

jhedberg, 2018-02-08:
Is this any different from https://github.com/zephyrproject-rtos/zephyr/issues/5486 ?

MariuszSkamra, 2018-02-12:
Yes, it is. Those tests are related to peripheral role, while in #5486 Zephyr is central

MariuszSkamra, 2018-02-12:
Tests are passing on QEMU with PTS dongle

nashif, 2018-02-20:
@MariuszSkamra  any updates?

PR #6417 (2018-03-02): Bluetooth: Workaround privacy feature issue
This fixes intercompatibility issues with controllers supporting
privacy feature.
Core Spec requires to use network privacy mode as a default when
peer device provides its IRK during bonding when LL Privacy is used,
which is the case for Zephyr. We've seen devices including PTS
which exchanges it's IRK but is not aware about network privacy
mode. This results in Zephyr not able do be reconnect to such bonded
devices.
This workaround sets device privacy mode to be able to reconnect
to such devices.

Fixes #4989
Fixes #5486
commit 777f9c8c (2018-03-02): Bluetooth: Workaround privacy feature issue

This fixes intercompatibility issues with controllers supporting
privacy feature.
Core Spec requires to use network privacy mode as a default when
peer device provides its IRK during bonding when LL Privacy is used,
which is the case for Zephyr. We've seen devices including PTS
which exchanges it's IRK but is not aware about network privacy
mode. This results in Zephyr not able do be reconnect to such bonded
devices.
This workaround sets device privacy mode to be able to reconnect
to such devices.

Fixes #4989
Fixes #5486
closed 2018-03-02 by carlescufi