zephyrproject-rtos/zephyr #5004: Implement RAW mode in KW41Z IEEE802514 driver

galak, 2017-11-15:
All the other drivers implement a RAW mode, enabled by a config like IEEE802154_MCR20A_RAW.  We should do this on KW41Z as well.  Eventually at least the config option should get refactored up a level.

galak, 2017-11-15:
@dleach02 can you take a look at this when you have a few cycles.  @jfischer-phytec-iot might be helpful to ask about exactly what RAW mode is doing/needs since he did something for MCR20A.

tbursztyka, 2017-11-16:
raw mode is useless but for wpan samples.

tbursztyka, 2017-11-16:
Actually, I think it's the good occasion to get rid of that awful hack that wpan introduced, I'll investigate.

dleach02, 2017-11-16:
I'm curious about what this raw mode is? I'll follow up with the Phytek guys to get some insight but my first instinct is similar to @tbursztyka. I prefer seeing systems using a common interface to the driver.

tbursztyka, 2017-11-16:
see PR #5018  

drivers will still need to implement a little bit of it: it should keep/write the FCS in the net_pkt. That's all.
(no more lqi written in the beginning etc... that was linux / wpan specific.)

PR #5018 (2017-11-22): net/ieee802154: Make RAW mode generic
- Renaming NET_L2_RAW_CHANNEL to NET_RAW_MODE
- Create a generic IEEE 802.15.4 raw mode for drivers
- Modify the IEEE 802.15.4 drivers so it passes the packet unmodified,
up to code using that mode to apply the necessary changes on the
received net_pkt according to their needs
- Modify wpanusb/wpan_serial relevantly

Fixes #5004
commit b52c0f24 (2017-11-22): net/ieee802154: Make RAW mode generic

- Renaming NET_L2_RAW_CHANNEL to NET_RAW_MODE
- Create a generic IEEE 802.15.4 raw mode for drivers
- Modify the IEEE 802.15.4 drivers so it passes the packet unmodified,
up to code using that mode to apply the necessary changes on the
received net_pkt according to their needs
- Modify wpanusb/wpan_serial relevantly

Fixes #5004
closed 2017-11-22 by nashif