zephyrproject-rtos/zephyr #5008: system call headers are not properly regenerated in CMake on incremental builds

andrewboie, 2017-11-15:
Steps to reproduce:

1) Do a build
2) Make a change to the function prototype of an API marked with __syscall
3) Build again

Observe that the associated header in <out>/include/generated/syscalls/ is not updated as the script was never re-run.

This worked in the old build system, there was a filechk_ rule for gen_syscalls.py that *always* ran the script, the generated files for syscall headers were not touched if they were determined to be the same as a previous run. See logic in gen_syscalls.py main() and the old build system logic that invoked gen_syscalls.py

Its not clear to me from looking at cmake/app/boillerplate.cmake under what conditions gen_syscalls.py is re-run on an incremental build (I don't understand cmake very well)

SebastianBoe, 2017-11-17:
Okay, I've tried many different methods, and I have a proposal that I
hope you can OK @andrewboie .

In any build system you have to create a DAG of dependencies between
files. In KBuild the DAG looked like this:

NULL (always out-of-date) -> generated syscalls -> Pretty much any object file in a Zephyr build

With a DAG like that you would think that KBuild would have to do a
full rebuild every time Make was invoked. But it didn't because it is
possible to "hack" make into supporting dependencies based on content
change instead of timestamps.

With CMake on the other hand it is impossible[0] to have dependencies
based on content changes. So to prevent the entire DAG from
becoming out-of-date on every make invocation we need a different DAG.

# Proposal

My proposal would be to follow what the Linux kernel does, have a
single header file that declares all syscalls.
https://github.com/torvalds/linux/blob/master/include/linux/syscalls.h
The DAG would then look like this:

syscalls.h -> generated syscalls -> Pretty much any object file in a Zephyr build

This is solution is simple, it makes sense from a build system point
of view and therefore has very good performance.

# Alternatives

A)
Patch, fork, and distribute a version of CMake that supports
checksum-based dependencies for custom targets. AFAICT this would
require too much effort.

B)
Investigate any possible hacks. Although not officially supported by
CMake, AFAICT the only reason that we are not able to emulate KBuild's
behaviour is that CMake is touching the custom targets. If we could
find out how to make it stop doing that we would avoid a timestamp
change and we would achieve content based dependencies. Looking at the
source code, the offending mechanism is:
https://github.com/Kitware/CMake/blob/master/Source/cmMakefileTargetGenerator.cxx#L955
so perhaps making the target SYMBOLIC would work (It didn't when I
tried quickly, but I can't explain why that wouldn't work).

C)
Maintain a list of headerfiles that contain syscalls like this:

set(syscall_header_files
    adc.h
    kernel.h
    [...]  # etc.
)

in the build system and create a DAG that looks like this:

[syscall_header_files] -> generated syscalls -> most object files

This would work pretty well I think, but is prone to users forgetting
to update the list.

D)

Run gen_syscalls.py at Configure time to determine what source files
have syscalls in them when CMake is invoked and then create a DAG like
this:

[syscall_header_files when CMake is invoked] -> generated syscalls -> most object files

This is also prone to user error. It must be documented to users that
if they introduce a new syscall then they MUST reconfigure (Re-run
CMake) to not corrupt the build.

I don't like D) because users should feel safe in calling "make", if
it is difficult to determine when to call "make" and when to call
"cmake" users will end up always doing clean builds and they
experience very poor performance. But if the proposal is denied
then this is the next best thing.

[0] I obviously can't prove a negative, but I have exhausted all my
ideas for hacking this support into CMake. Mailing list discussion
about checksum dependencies:
https://public.kitware.com/pipermail/cmake/2012-August/051555.html

SebastianBoe, 2017-11-20:
I did not like the proposed solution, so I tried a different approach and succeeded with what is alternative B) above in PR #5071

closed 2017-12-04 by andrewboie