zephyrproject-rtos/zephyr #5009: cmake creates too many build artifacts

andrewboie, 2017-11-15:
As a baseline, any Zephyr build needs to produce a .elf file for the kernel.
Depending on the target hardware requirements, this .elf file *may* need to be additionally converted into another binary type to support device provisioning.

Examples:

- Nios II Altera MAX10 needs the kernel converted to IHEX format
- Many targets (but not the QEMU targets) need the elf stripped and turned into BIN format
- RiscV32 needs the ELF turned into a S19 format file
- IIRC, ESP32 and all the QEMU targets take the ELF file directly and don't need additional build artifacts

What CMake seems to be doing is generating all these different output file types unnecessarily for all board targets.

What we need is at a board or SOC level, have some Kconfig which tells the build system what output artifact types are necessary for that board.
For targets like QEMU targets, no additional artifacts need to be created at all; the ELF is the final build product.

The scripts for "make flash" for our boards should provide clues on what particular build artifacts are needed for any given board.

lpereira, 2017-11-15:
Just a clarification: ESP32 takes the striped ELF and turns it into a format that's specific to ESP32 (it's the format that the first stage bootloader, part of the SoC mask ROM, understands).  The .bin file generated in the output directory is not used by the flasher tool; the flashing script will overwrite the BIN format with this proprietary format prior to flashing.

SebastianBoe, 2017-11-20:
Modifying the build scripts is pretty straightforward.
https://github.com/SebastianBoe/zephyr/commit/4da2f76a87c9465d6e300943a0fcfa90ac6b8eac

Creating a user-friendly Kconfig is a bit more tricky, I'll give it a go later on unless someone has any ideas themselves. I imagine we might want a menu for the configuring the build system. Perhaps the menu "Compiler Options" could be sub-menu in this "Build system options" menu.

nashif, 2017-11-22:
@SebastianBoe can you submit this as a PR, I am looking into making this configurable using the board yaml files which will be able to specify which output binaries they need and will support. This will be done in a layered approach, so we can specify this for an architecture or a group of boards as well.

SebastianBoe, 2017-11-22:
Done: https://github.com/zephyrproject-rtos/zephyr/pull/5112

PR #5114 (2017-11-23): generate artifacts based on Kconfig defined by SoC/Board
Not all boards require the various binary formats zephyr generates. So be
selective based on the arch, SoC or board and only geenrate the binaries
actually needed.

Fixes #5009
commit 1f1143ac (2017-11-23): build: use kconfig to select generated artifacts

Not all boards require the various binary formats zephyr generates. So
be selective based on the arch, SoC or board and only geenrate the
binaries actually needed.

Fixes #5009
closed 2017-11-23 by nashif