zephyrproject-rtos/zephyr #5017: Genuino/Arduino 101 Sensor Sample: Gyro Device not found

dhavalpanchalispl, 2017-11-16:
**Board:** Genuino 101 
**Zephyr SDK:** Latest version 0.9.2 

I flashed both ARC and X86 cores successfully using these samples,

**ARC Core:**
cd $ZEPHYR_BASE/samples/sensor/bmi160
mkdir -p build/arduino_101_sss && cd build/arduino_101_sss
cmake -DBOARD=arduino_101_sss ../..
make
make flash

**X86 Core:**
cd $ZEPHYR_BASE/tests/booting/stub
mkdir -p build/arduino_101 && cd build/arduino_101
cmake -DBOARD=arduino_101 ../..
make
make flash

When we run the BMI160 sample, it shows this in console,
IMU: Binding...
Gyro: Device not found.

My issue is similar to [this](https://github.com/zephyrproject-rtos/zephyr/issues/3157). Call to **device_get_binding(CONFIG_BMI160_NAME) always returns NULL**.

nashif, 2017-11-16:
works just fine:

```
***** BOOTING ZEPHYR OS v1.9.99 - BUILD: Nov 16 2017 11:53:53 *****
IMU: Binding...
Testing the polling mode.
Gyro (rad/s): X=-0.006118, Y=0.008379, Z=0.009975
Acc (m/s^2): X=-0.751716, Y=-1.067724, Z=9.303084
Temperature (Celsius): 29.907761
```

nashif, 2018-02-08:
cant reproduce, closing

closed 2018-02-08 by nashif