zephyrproject-rtos/zephyr #5019: tests/kernel/mem_protect/stackprot : input string is long stack overflow

rrahulkx, 2017-11-16:
The procedure followed to re-produce is -

Zephyr Commit ID:-
036dde2201ad60f2f1df2c73b62031a6967c629d
Config file is attached to this Report:
[kernel_mem_protect_stackprot_config.txt](https://github.com/zephyrproject-rtos/zephyr/files/1478228/kernel_mem_protect_stackprot_config.txt)

Build Procedure-

Cmake Build procedure -

$cd $ZEPHYR_BASE/tests/kernel/mem_protect/stackprot
 $mkdir build && cd build
 $cmake –DBOARD=quark_se_c1000_devboard..
 $make flash

Detailed description
====================
build passed: 'mkdir -p outdir-aetj-ijnd-quark_se_c1000_ss_devboard; rm -f outdir-aetj-ijnd-quark_se_c1000_ss_devboard/.config'

build passed: 'cmake -DPREBUILT_HOST_TOOLS= 
-DBOARD=quark_se_c1000_ss_devboard 
-DARCH=arc -DEXTRA_CPPFLAGS="-DTC_RUNID=:ijnd" 
-DEXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations" 
-DEXTRA_AFLAGS=-Wa,--fatal-warnings -DEXTRA_LDFLAGS=--fatal-warnings  -B"outdir-aetj-ijnd-quark_se_c1000_ss_devboard" -H"zephyr/tests/kernel/mem_protect/stackprot"'

build passed: 'mkdir -p outdir-aetj-ijnd-quark_se_c1000_devboard; rm -f outdir-aetj-ijnd-quark_se_c1000_devboard/.config'
build passed: 'cmake -DPREBUILT_HOST_TOOLS=zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/bin -DBOARD=quark_se_c1000_devboard -DARCH=x86
 -DEXTRA_CPPFLAGS="-DTC_RUNID=:ijnd"
 -DEXTRA_CFLAGS="-Werror -Wno-error=deprecated-declarations"
 -DEXTRA_AFLAGS=-Wa,--fatal-warnings
 -DEXTRA_LDFLAGS=--fatal-warnings  -
B"outdir-aetj-ijnd-quark_se_c1000_devboard" 
-H"zephyr/tests/booting/stub"'

build passed

deploy passed

Reset

console output: starting test - Test Stack Protection Canary
console output:
console output: Starts main
console output: Starts alternate_thread
console output: alternate_thread: Input string is too long and stack overflowed!
console output:
console output: alternate_thread: Stack ok
console output: alternate_thread: Stack ok
console output: alternate_thread: Stack ok
console output: alternate_thread: Stack ok
console output: alternate_thread: Stack ok
console output: alternate_thread: Stack ok
console output: Exception vector: 0x3, cause code: 0x6, parameter 0x0
console output: Address 0x206f6f74
console output: Current thread ID = 0xa8000450
console output: Faulting instruction address = 0x206f6f74
console output: Fatal fault in thread 0xa8000450! Aborting.
console output: FAIL - main.
console output:

andrewboie, 2017-11-16:
investigating
Hoping this is reproducible on arduino_101_ss

andrewboie, 2017-11-28:
The compiler doesn't seem to be generating the stack canary code on ARC.
Maybe the compiler doesn't support it?

andrewboie, 2017-11-28:
```
Performing C SOURCE FILE Test check_fstack_protector_all failed with the following output:
Change Dir: /projects/zephyr/tests/kernel/mem_protect/stackprot/out/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/gmake" "cmTC_b2783/fast"
/usr/bin/gmake -f CMakeFiles/cmTC_b2783.dir/build.make CMakeFiles/cmTC_b2783.dir/build
gmake[1]: Entering directory '/projects/zephyr/tests/kernel/mem_protect/stackprot/out/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_b2783.dir/src.c.obj
/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arc-zephyr-elf/arc-zephyr-elf-gcc   -Dcheck_fstack_protector_all -nostartfiles -nostdlib -isystem /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arc-zephyr-elf/../../lib/arc-zephyr-elf/gcc/arc-zephyr-elf/6.2.1/include -isystem /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arc-zephyr-elf/../../lib/arc-zephyr-elf/gcc/arc-zephyr-elf/6.2.1/include-fixed -Wl,--unresolved-symbols=ignore-in-object-files   -fstack-protector-all -o CMakeFiles/cmTC_b2783.dir/src.c.obj   -c /projects/zephyr/tests/kernel/mem_protect/stackprot/out/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_b2783
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_b2783.dir/link.txt --verbose=1
/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arc-zephyr-elf/arc-zephyr-elf-gcc  -Dcheck_fstack_protector_all -nostartfiles -nostdlib -isystem /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arc-zephyr-elf/../../lib/arc-zephyr-elf/gcc/arc-zephyr-elf/6.2.1/include -isystem /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arc-zephyr-elf/../../lib/arc-zephyr-elf/gcc/arc-zephyr-elf/6.2.1/include-fixed -Wl,--unresolved-symbols=ignore-in-object-files    CMakeFiles/cmTC_b2783.dir/src.c.obj  -o cmTC_b2783 
/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arc-zephyr-elf/../../libexec/arc-zephyr-elf/gcc/arc-zephyr-elf/6.2.1/real-ld: warning: cannot find entry symbol __start; defaulting to 0000000000000124
CMakeFiles/cmTC_b2783.dir/src.c.obj: In function `main':
src.c:(.text+0x10): undefined reference to `__stack_chk_guard'
src.c:(.text+0x24): undefined reference to `__stack_chk_guard'
src.c:(.text+0x2c): undefined reference to `__stack_chk_fail'
collect2: error: ld returned 1 exit status
gmake[1]: *** [CMakeFiles/cmTC_b2783.dir/build.make:98: cmTC_b2783] Error 1
gmake[1]: Leaving directory '/projects/zephyr/tests/kernel/mem_protect/stackprot/out/CMakeFiles/CMakeTmp'
gmake: *** [Makefile:126: cmTC_b2783/fast] Error 2

Source file was:
int main(void) { return 0; }
```

andrewboie, 2017-11-28:
If I manually pass -fstack-protector-all it works.
The build system rejects the option because the test code fails due to linker error.
I'm mystified why this doesn't fail on other arches, the stack protector feature requires that the __stack_chk_guard and __stack_chk_fail symbols are defined, I don't see how this could work on other arches since we are building free-standing binaries.

andrewboie, 2017-11-28:
I could investigate further why this only fails on ARC, but I think a different approach is needed.

It seems we do these checks for ALL c compiler options, anytime zephyr_cc_option() is used.
Any option that doesn't pass the test, is dropped from cflags when the kernel is built.
In this case, I think this is not ideal. I would much rather see the build break if someone tries to enable the stack canaries, and the compiler doesn't support it. Because what happens now is that if someone enables this option, and there is no support, the build will succeed but there are no actual stack canaries in place, and unless the user is paying close attention to the cmake test output they will have no idea.

@SebastianBoe can we add a zephyr_cc_option_nocheck() or something like that which just adds an option unconditionally?

PR #5185 (2017-11-28): add -fstack-protector-all without checks 
- kernel: stack: add -fstack-protector-all without checks 
- cmake: add zephyr_cc_option_nocheck

Fixes #5019
commit eb88bf2e (2017-11-28): kernel: stack: add -fstack-protector-all without checks

Fixes #5019
closed 2017-11-28 by nashif