zephyrproject-rtos/zephyr #5025: arduino_due not generating proper config with cmake (crash)

Zarkopafilis, 2017-11-16:
I am trying to build the sample 'hello_world' app, after having properly installed everything according to the guide (msys on windows + gccarmemb) and I get the following error:

`
$(inside /build) cmake -DBOARD=arduino_due ..
-- Selected BOARD arduino_due
-- Generating zephyr/include/generated/autoconf.h
*
* Restart config...
*
*
* General Kernel Options
*
Multi-threading (MULTITHREADING) [Y/n/?] y
  Number of coop priorities (NUM_COOP_PRIORITIES) [16] 16
  Number of preemptible priorities (NUM_PREEMPT_PRIORITIES) [15] 15
Priority of initialization/main thread (MAIN_THREAD_PRIORITY) [0] 0
Priority inheritance ceiling (PRIORITY_CEILING) [0] 0
Size of stack for initialization and main thread (MAIN_STACK_SIZE) [1024] 1024
Size of stack for idle thread (IDLE_STACK_SIZE) [256] 256
ISR and initialization stack size (in bytes) (ISR_STACK_SIZE) [2048] 2048
Thread stack info (THREAD_STACK_INFO) [N/y/?] n
Thread custom data (THREAD_CUSTOM_DATA) [N/y/?] n
Enable errno support (ERRNO) [Y/n/?] y
Split kernel and application memory (APPLICATION_MEMORY) [N/y/?] n
System tick frequency (in ticks/second) (SYS_CLOCK_TICKS_PER_SEC) [100] 100
System clock's h/w timer frequency (SYS_CLOCK_HW_CYCLES_PER_SEC) [] (NEW) aborted!

Console input/output is redirected. Run 'make oldconfig' to update configuration.

/c/Users/Zarkopafilis/Documents/uop/due/build/zephyr/.config:79:warning: symbol value '' invalid for SYS_CLOCK_HW_CYCLES_PER_SEC
CMake Error at /home/Zarkopafilis/zephyr/cmake/gcc-m-cpu.cmake:20 (message):
  Expected CONFIG_CPU_CORTEX_x to be defined
Call Stack (most recent call first):
  /home/Zarkopafilis/zephyr/cmake/toolchain-gcc.cmake:52 (include)
  /home/Zarkopafilis/zephyr/cmake/toolchain.cmake:16 (include)
  /home/Zarkopafilis/zephyr/cmake/app/boilerplate.cmake:218 (include)
  CMakeLists.txt:1 (include)

-- Configuring incomplete, errors occurred!
`

I could try to set SYS_CLOCK_HW_CYCLES_PER_SEC to 84MHz <=> 84000000 Hz to remove the warning but I guess that is wrong and doesn't need to be set.

For the CONFIG_CPU_CORTEX_x, I don't know what to do. I tried setting it to M3 or SAM3X8E but the config file would rewrite upon the cmake command.

 No makefile or ninja files are generated of course...

SebastianBoe, 2017-11-17:
Was not able to reproduce with this environment:

OS: Ubuntu
Toolchain: Zephyr SDK
sample: samples/hello_world
BOARD: arduino_due
git revision: b05ba6b531ba310dc5f3190c846177f52f805b66

SebastianBoe, 2017-11-17:
I have seen this before with a non-clean build directory. Can you confirm that you can reproduce this with a clean build?

SebastianBoe, 2017-11-17:
Was not able to reproduce with gccarmemb either.

Will try to reproduce on Windows after you have confirmed it affects clean builds on git revision b05ba6b.

closed 2017-11-17 by Zarkopafilis