zephyrproject-rtos/zephyr #5040: Bluetooth mesh API documentation incomplete

dbkinder, 2017-11-17:
@jhedberg noted that the generated API documentation for Bluetooth Mesh (found at http://docs.zephyrproject.org/api/bluetooth.html#mesh-profile) is missing material.  A quick look shows only API material from bluetooth/mesh/main.h is showing up.  

Recommend folks read the header files for now while we investigate and fix.

dbkinder, 2017-11-17:
Verified doxygen comments and processing is ok, so it looks like a sphinx/breathe issue...

nashif, 2017-11-18:
see http://docs.zephyrproject.org/apidoc/latest/dd/d6d/group__bt__mesh.html

dbkinder, 2017-11-19:
Wait for another cycle of the nightly doc build to verify the subgroups of Mesh in the http://docs.zephyrproject.org/api/bluetooth.html API doc are shown and we can close this issue.

closed 2017-11-28 by dbkinder