zephyrproject-rtos/zephyr #5047: document error: getting_started.rst

zhaozhongchen, 2017-11-19:
In the section **Building without the Zephyr SDK**

is there lost **cd $ZEPHYR_BASE** before *make* ?

```
$ cd $ZEPHYR_BASE
$ mkdir build && cd build
$ cmake $ZEPHYR_BASE/scripts
$ make
$ echo "export PATH=$PWD/kconfig:\$PATH" >> $HOME/.zephyrrc
$ source $ZEPHYR_BASE/zephyr-env.sh
```

dbkinder, 2017-11-19:
I believe it is correct as written; make is supposed to run in the ``build`` directory, right @SebastianBoe

zhaozhongchen, 2017-11-19:
@dbkinder But there is no such file named *Makefile* in build directory.

zhaozhongchen, 2017-11-19:
@dbkinder  sorry, it's my fault. the right way is **cmake $ZEPHYR_BASE/scripts**, but i forgot to type **scripts**

dbkinder, 2017-11-19:
OK.  We'll close this issue then.

closed 2017-11-19 by dbkinder