zephyrproject-rtos/zephyr #5087: Samples/bluetooth: Failed to connect with eddystone sample app on arduino_101.

vikramx2112, 2017-11-21:
Eddystone sample app is designed to advertise eddystone pkts and waiting for connection for 30sec after boot. If any connect request comes within 30 sec, then connection is successful. In zephyr version v1.9-branch, connection is getting successful, but with latest version connection failed on arduino_101.

Error logs:
console: ***** BOOTING ZEPHYR OS v1.10.0-rc1 (delayed boot 1000ms) *****
[bt] [WRN] hci_vs_init: Vendor HCI extensions not available
[bt] [WRN] set_static_addr: Read Static Addresses command not available
[bt] [WRN] set_static_addr: Using temporary static random address
[bt] [INF] show_dev_info: Identity: e7:8c:6c:22:d0:23 (random)
[bt] [INF] show_dev_info: HCI: version 5.0 (0x09) revision 0x0000, manufacturer 0xffff
[bt] [INF] show_dev_info: LMP: version 5.0 (0x09) subver 0xffff
Bluetooth initialized
Configuration mode: waiting connections...
[bt] [ERR] hci_acl: Unable to find conn for handle 
[bt] [WRN] report_completed_packet: Unable to look up conn with id 0xff
Switching to Beacon mode.
Advertising failed to start (err -69)

Tested zephyr version:
Latest version: cb9390c

How to reproduce.

1. Take one zephyr ble target(arduino_101) and ble dongle connected with Linux machine. And placed both BLE devices near to each other.
2. On zephyr target: Install eddystone sample app($ZEPHYR_BASE/samples/bluetooth/eddystone).
3. Check on console and waiting for "Configuration mode: waiting connections...".
4. Run bluetoothctl tool on linux machine(where BLE dongle is connected).
5. Run below commands to connect BLE dongle to arduino_101.
	$ bluetoothctl
	$ agent on
	$ default-agent
	$ remove * 		# remove all scanned devices from cache.
	$ scan on		# scan for eddystone advertisment pkt.
	# waiting for eddystone advertise pkt.
	$ scan off
	$ connect $MAC_ADDR_A101	# $MAC_ADDR_A101 is a BLE mac addr of a101.
# Make sure above steps must be done before 30sec. As after that eddystone sample app will start sending non-connectable beacons.		

Expected result:

1. Zephyr target and ble dongle must be connected.

Actual result:

1. Zephyr target and ble dongle is not connecting and getting error of "Unable to find conn for handle".

jhedberg, 2017-11-24:
Could you also flash your controller with the latest Zephyr version? From the logs its evident that you have some older one.

vikramx2112, 2017-11-27:
Thanks @jhedberg, with latest BLE FW, test is passing. So closing the issue thread.

closed 2017-11-27 by vikramx2112