zephyrproject-rtos/zephyr #5091: MPU fault at drivers/flash/soc_flash_nrf5.c:493 with NFFS enabled on nrf52840_pca10056

jhedberg, 2017-11-21:
```***** BOOTING ZEPHYR OS v1.10.0-rc1 - BUILD: Nov 21 2017 06:55:54 *****
shell> ***** MPU FAULT *****
  Executing thread ID (thread): 0x20003d78
  Faulting instruction address:  0x4568
  Data Access Violation
  Address: 0xfc000
Fatal fault in essential thread! Spinning...
```

The instruction address 0x4568 resolves to `drivers/flash/soc_flash_nrf5.c:493`

Happens with the following configuration using them mesh_shell app:

```diff --git a/tests/bluetooth/mesh_shell/prj.conf b/tests/bluetooth/mesh_shell/prj.conf
index 13ab163329..21a78f04db 100644
--- a/tests/bluetooth/mesh_shell/prj.conf
+++ b/tests/bluetooth/mesh_shell/prj.conf
@@ -1,3 +1,11 @@
+CONFIG_FLASH=y
+CONFIG_FILE_SYSTEM=y
+CONFIG_SOC_FLASH_NRF5=y
+CONFIG_FILE_SYSTEM_NFFS=y
+CONFIG_SOC_FLASH_NRF5_DEV_NAME="NRF52_FLASH"
+CONFIG_FS_NFFS_FLASH_DEV_NAME="NRF52_FLASH"
+CONFIG_FILE_SYSTEM_SHELL=y
+
 #CONFIG_INIT_STACKS=y
 CONFIG_USERSPACE=n
 CONFIG_MAIN_STACK_SIZE=256
```

jhedberg, 2017-11-21:
Just verified that this also happens with samples/hello_world:

```***** BOOTING ZEPHYR OS v1.10.0-rc1 - BUILD: Nov 21 2017 07:52:30 *****
***** MPU FAULT *****
  Executing thread ID (thread): 0x20001964
  Faulting instruction address:  0x19c8
  Data Access Violation
  Address: 0xfc000
Fatal fault in essential thread! Spinning...
```

And instruction address 0x19c8 resolves again to `drivers/flash/soc_flash_nrf5.c:493`

```$ git diff samples/hello_world/
diff --git a/samples/hello_world/prj.conf b/samples/hello_world/prj.conf
index b2a4ba5910..613ade836c 100644
--- a/samples/hello_world/prj.conf
+++ b/samples/hello_world/prj.conf
@@ -1 +1,7 @@
 # nothing here
+CONFIG_FLASH=y
+CONFIG_FILE_SYSTEM=y
+CONFIG_SOC_FLASH_NRF5=y
+CONFIG_FILE_SYSTEM_NFFS=y
+CONFIG_SOC_FLASH_NRF5_DEV_NAME="NRF52_FLASH"
+CONFIG_FS_NFFS_FLASH_DEV_NAME="NRF52_FLASH"
```

jhedberg, 2017-11-21:
`CONFIG_MPU_ALLOW_FLASH_WRITE=y` fixes this. It seems to me this option should be enabled by default, and if it's disabled we should give an understandable error to the user saying this option needs to be enabled, in case the FS isn't formatted yet.

nvlsianpu, 2017-11-24:
I open related 'TBD' task https://github.com/zephyrproject-rtos/zephyr/issues/5151
I think it will be better to create separate MPU flash region for data storage.

nvlsianpu, 2018-01-30:
CONFIG_MPU_ALLOW_FLASH_WRITE is selected by NFFS enabling - then this issue is fixed.

closed 2018-01-30 by nvlsianpu