zephyrproject-rtos/zephyr #5101: LwM2M: device hang after requesting a firmware update to a loopback device IP

rbtchc, 2017-11-22:
Tested against commit b678895a6dadff7a10cb5ff6927ae74cc49fbae0

Issue a PACKAGE URI firmware update (/5/0/1) w/ URI set as _coap://127.0.0.1:7783/large_ to start the firmware update process. The device will hang shortly after receive the request from server.
Issue is reproducible on both Qemu_x86 and FRDM K64F

Debug log as below
> [net/conn] [DBG] net_conn_input: (0x00403660): Check UDP listener for pkt 0x00406060 src port 5683 dst port 37042 family 10 chksum 0x1186 len 48
> [net/conn] [DBG] net_conn_input: (0x00403660): [0] match found cb 0x0000f302 ud 0x00400080 rank 0x33
> [net/ctx] [DBG] packet_received: (0x00403660): Set appdata 0x0040564e to len 40 (total 88)
> [lib/lwm2m_engine] [DBG] lwm2m_udp_receive: checking for reply from [2001:db8::2]
> [lib/lwm2m_engine] [DBG] handle_request: No accept option given. Assume OMA TLV.
> [net/app] [DBG] _net_app_select_net_ctx_debug: (0x00403660): Selecting 0x00403480 net_ctx (net_app_get_net_pkt():960)
> [lwm2m_obj_firmware] [DBG] package_uri_write_cb: PACKAGE_URI WRITE: coap://127.0.0.1:7783/large
> [lwm2m_obj_firmware] [DBG] lwm2m_firmware_set_update_state: Update state = 0
> [lwm2m_obj_firmware] [DBG] lwm2m_firmware_set_update_result: Update result = 0
> [lwm2m_obj_firmware] [DBG] lwm2m_firmware_set_update_state: Update state = 1
> [net/ipv6] [DBG] net_ipv6_prepare_for_send: (0x00403660): No default route to 2001:db8::2
> [net/ipv6] [DBG] net_ipv6_prepare_for_send: (0x00403660): Neighbor lookup 0x00422260 (0) iface 0x00422ca0 addr 2001:db8::2 state stale
> [net/ipv6] [DBG] net_ipv6_prepare_for_send: (0x00403660): Neighbor 0x00422260 addr D6:EF:DF:AA:6D:81
> [net/ipv6] [DBG] ipv6_nbr_set_state: (0x00403660): nbr 0x00422260 stale -> delay
> [net/ethernet] [DBG] ethernet_send: (0x00403660): src 00:00:5E:00:53:4C dst D6:EF:DF:AA:6D:81 type 0x86dd len 53
> [net/core] [DBG] processing_data: (0x00403660): Consumed pkt 0x00406060
> [net/if] [DBG] net_if_tx: (0x00403720): Processing (pkt 0x00405de0, data len 53) network packet
> [net/if] [DBG] net_if_tx: (0x00403720): Calling context send cb 0x00403480 token 0x00000000 status 0
> [net/ctx] [DBG] net_context_bind: (0x00404330): Context 0x004034dc binding to UDP 192.0.2.1:59282 iface 0x00422ca0
> [net/app] [DBG] _net_app_print_info: (0x00404330): net app connect 192.0.2.1:59282 -> 127.0.0.1:7783
> [lwm2m_obj_firmware_pull] [INF] firmware_transfer: Connecting to server 127.0.0.1:7783/large, port 7783
> [net/app] [DBG] _net_app_select_net_ctx_debug: (0x00404330): Selecting 0x004034dc net_ctx (net_app_connect():664)
> [net/conn] [DBG] net_conn_register: (0x00404330): [1/2/17/0x33] remote 0x004034f0/127.0.0.1/7783 local 0x004212f4/192.0.2.1/59282 cb 0x0000f302 ud 0x00402c20
> [net/app] [DBG] _net_app_select_net_ctx_debug: (0x00404330): Selecting 0x004034dc net_ctx (net_app_get_net_pkt():960)
> [net/core] [DBG] net_send_data: (0x00404330): Loopback pkt 0x00405de0 back to us
> [net/ipv4] [DBG] net_ipv4_process_pkt: (0x00404330): IPv4 packet received from 127.0.0.1 to 192.0.2.1
> [net/conn] [DBG] net_conn_input: (0x00404330): Check UDP listener for pkt 0x00405de0 src port 59282 dst port 7783 family 2 chksum 0xee7e len 49
> [net/conn] [DBG] net_conn_input: (0x00404330): No match found.
> [net/icmpv4] [DBG] net_icmpv4_send_error: (0x00404330): Sending ICMPv4 Error Message type 3 code 3 from 192.0.2.1 to 127.0.0.1
> [net/core] [DBG] net_send_data: (0x00404330): Loopback pkt 0x00405d60 back to us
> [net/ipv4] [DBG] net_ipv4_process_pkt: (0x00404330): IPv4 packet received from 127.0.0.1 to 192.0.2.1
> [net/ipv4] [DBG] process_icmpv4_pkt: (0x00404330): ICMPv4 packet received type 3 code 3
> [net/core] [DBG] processing_data: (0x00404330): Dropping pkt 0x00405d60
> [net/core] [DBG] processing_data: (0x00404330): Dropping pkt 0x00405de0
> 
>

rbtchc, 2017-12-07:
Digging a bit more.
It seems that the packet sent to loopback device will be unreferenced immediately when the packet.
lwm2m_send_message() doesn't get a chance to call coap_pending_cycle() to increase the reference count to avoid the packet from being freed which result in the device hang.

If I requested a firmware pull from an IP other than loopback, it seems that an ARP packet is generated and issued first for the IP resolving and then the firmware pull request packet.
The timing seems long enough for lwm2m_send_message() to call coap_pending_message() to increase the reference count to protect the packet from being released.

rbtchc, 2017-12-07:
Another issue is that, even if the packet is correctly protected. The loop device will swap the src/dst IP.
For protocol like coap which do the retransmission at the application layer.
The packet it reserved is actually modified after the first attempt.

pfalcon, 2017-12-07:
> It seems that the packet sent to loopback device will be unreferenced immediately when the packet.

There's a missing word apparently.

pfalcon, 2017-12-07:
> For protocol like coap which do the retransmission at the application layer.

Ditto.

pfalcon, 2017-12-07:
It's not cleat what's the problem is being reported here? Do you report that if you randomly try to use loopback (which loopback exactly?) that COAP fails randomly? Or do you use loopback on purpose? Which? What are your expectations?

(Asking as the author of https://github.com/zephyrproject-rtos/zephyr/pull/4461 , though I'm not even sure you use *that* loopback).

rbtchc, 2017-12-07:
Let me clarify.
LwM2M server can send a URL to LwM2M client to ask it to perform a firmware pull.
I accidentally enter an URL "coap://127.0.0.1:7783/large" to the client.
Client then tries to send an UDP packet to 127.0.0.1, port 7783 which will be looped back to the client and the connection manager will drop the packet immediately for no one is listening to the port 7783 at client (NOTE: at this point, the packet reference is decreased to 0).
Later on, the LwM2M client realized that it did not receive any confirmation packet at CoAP layer. It will try to retransmit the packet again. But of course, the packet is already been released and the data are corrupted.

rbtchc, 2017-12-07:
I think the issues is that LwM2M should protect the packet from being released by the lower layers.
And I'm expecting the packet stays the same even if it sends through the loopback device.

pfalcon, 2017-12-07:
> I accidentally enter an URL "coap://127.0.0.1:7783/large" to the client.

That explains it, so you by a chance tried that and saw it break "spectacularly".

I don't think the problem is LwM2M related though, or should be handled on LwM2M level. It likely affects other UDP (or maybe not just UDP based) services.

Besides the patch linked above, you may also look at https://github.com/zephyrproject-rtos/zephyr/pull/4956 as another patch touched those matters.

And if you'd like to "address" it somehow quick, you might try to disable loopback handling and see how it behaves then.

rbtchc, 2017-12-08:
The CONFIG_NET_LOOPBACK is unset at my case.

rbtchc, 2018-01-16:
Test against 9851b1ac2188f791741b2f2b2b7ea5547a6cd2a1, the issue is fixed. Therefore, close it.

closed 2018-01-16 by rbtchc