zephyrproject-rtos/zephyr #5109: yaml: fix key/value syntax to 'mapping' instead of 'series'

erwango, 2017-11-22:
According to yaml syntaxic rules, '- attribute' should be used for series (list of single element)
and 'attribute' (w/o hyphen) should be used for 'mapping' (key/value couple),
When treated by yaml python library, yaml mappings are converted to python type 'dict'
while series are converted to python type 'list'.

In some yaml files, we're using hyphen before key/value element.
They are then converted to 'list' of key/value element instead of 'dict' which is not ideal to deal with in python.

Rework yaml files to be syntaxically correct and do the modifications (which should be code simplification) in scripts that deal with the yaml files.

erwango, 2017-11-22:
Impacted:
-dts/bindings/\*/\*.yaml  -> fixed by #5076
-samples/\*/\*.yaml
-tests/\*/\*.yaml

galak, 2017-11-28:
Can we write a script or something to modify the yaml files?

nashif, 2017-12-05:
> Can we write a script or something to modify the yaml files?

i have a script already and will take  care of that.

commit 23f81eeb (2017-12-11): tests/samples: fixed yaml syntax

Use a map directory, avoid the list which makes parsing a bit
cumbersome.

Fixes #5109
closed 2017-12-11 by nashif