zephyrproject-rtos/zephyr #5138: dts: boards: provide generic dtsi file for 'generic' boards

erwango, 2017-11-23:
Some boards belonging to the same board family share common settings.
Being able to include provide a <generic_board>.dtsi file by each board adhering to the generic type would allow to factorize board description.

For instance st nucleo boards share:
-User Button (SW0), LD0 settings
-Console being routed to uart2
-uart2 pinmux and pinmuxing in general

These settings could be defined in a nucleo.dtsi file included by nucleo_xxxx.dts files.
Drawback of this proposal would be loss of readability of dts files as most information would be removed from board dts files.

psidhu, 2017-12-05:
I think device tree in zephyr could use a lot of work, this being one of them. I'm of the belief that all nucleo boards should have a single maintainer that can manage what you're mentioning. This goes for other 'groups' of boards as well.

Of course, someone else can do the work, but the maintainer should be responsible for keeping things sane.

erwango, 2017-12-06:
@psidhu : If you're interested into dt work, maybe you could have a look to following RFC and provide feedback:
https://github.com/zephyrproject-rtos/zephyr/pull/5021

erwango, 2019-09-19:
No interest in this for now. Closing

closed 2019-09-19 by erwango