zephyrproject-rtos/zephyr #5140: CMake migration regressed (changed behavior) of QEMU_PTY=1

pfalcon, 2017-11-23:
Turns out that older command `make run QEMU_PTY=1` doesn't work as expected (as it used to work before) - QEMU_PTY option isn't processed and QEMU doesn't run in PTY mode.

Instead, seems that the option needs to be specified to cmake invocation: `cmake ... -DQEMU_PTY=1`.

I'd argue this doesn't work as expected. The idea behind QEMU_PTY option (as far as I see it, granted, I picked it up for usage, not introduced it) is either run QEMU in interactive mode for human consumption, or in PTY mode to interface with tool (PTY mode effectively emulating running a tool over serial connection to a physical board). Switching between the two by means of complete rebuilding (via re-running cmake) is quite inconvenient.

This for example broke regression testing for MicroPython port (MicroPython runs its testsuite on a real board over serial; I used PTY mode to make it possible to run the testsuite against a QEMU "virtual board").

SebastianBoe, 2017-11-23:
This usability issue is of the same nature, and might have a similair solution, as https://github.com/zephyrproject-rtos/zephyr/issues/4957

pfalcon, 2017-11-23:
Thanks for the quick response and a pointer. It would be similar, if folks in #4957 wanted something like `make flash METHOD=jlink`, `make flash METHOD=pyocd`. But as they want separate make target, that indeed would be handled on cmake level. But make-level variables would need to be processed on make level, via make's builtin functions and/or shell statements, so we'd need to just generate corresponding code in cmake.

Here's a step towards that, taking a simpler case for starters: https://github.com/zephyrproject-rtos/zephyr/pull/5142

SebastianBoe, 2017-11-24:
You are correct pfalcon. This is a different use-case from #4957, because we can't create a new make target for every possible qemu_instance value.

In general we want to avoid using environment variables like we do in #5142, but for this use-case there does not exist any better solution.

SebastianBoe, 2017-11-24:
> because we can't create a new make target for every possible qemu_instance value.

Well, I guess we could create a dozen targets and require users to work within that subset of instance values. But we can deal with that use-case if it arises.

pfalcon, 2017-11-24:
@SebastianBoe :

> In general we want to avoid using environment variables like we do in #5142, but for this use-case there does not exist any better solution.

Per my clarification in https://github.com/zephyrproject-rtos/zephyr/pull/5142#discussion_r152957143 , this is not "environment variable", it's "make variable". (Make is flexible though and can take its variable from shell (process) environment. The point, it doesn't have too.)

(I'm sure you know all that, just phrase "environment variable" is confusing, and in the context you use it, there seems to be prejudice against "using environment variable". To not dig into that, I point that these are *make* variables, and people definitely want to use them, like they did before CMake came. Actually, some people want CMake to be completely invisible and not leak into their daily workflow.)

SebastianBoe, 2017-11-24:
It doesn't have to, but it can, which is unfortunate.

SebastianBoe, 2017-11-29:
Resolved by #5154