zephyrproject-rtos/zephyr #5143: Cmake ignores setting CONFIG_ETH_MCUX_0=n in prj.conf

vakulgarg, 2017-11-24:
I am compiling samples/bluetooth/ipsp for frdm_k64f board with following modifications in prj.conf.
The following two lines are added.

+CONFIG_BT=y
+CONFIG_NET_L2_BT=y
+CONFIG_ETH_MCUX_0=n
+CONFIG_NET_L2_ETHERNET=n

In the  zephyr configuration generated ipsp/build/zephyr/.config, I still see CONFIG_ETH_MCUX_0=y.
Thus CONFIG_ETH_MCUX_0 does not get turned off.

SebastianBoe, 2017-11-24:
That is weird, I see that the board's kconfig defaults it to y.

config ETH_MCUX_0
	def_bool y

But CONF_FILE is passed to merge_config.sh after the board's kconfig, so merge_config should override the board's settings. I can't explain this behaviour.

vakulgarg, 2017-11-25:
Thanks for looking into this. Does the problem reproduce at your end?

Regards
Vakul

nashif, 2017-11-25:
this config is being set in the board when networking is enabled, looks like it is not overridable, which is a problem.

carlescufi, 2018-02-06:
I tried to reproduce this but `CONFIG_ETH_MCUX_0` is disabled in my `.config`. Will close this unless someone can reproduce this.

nashif, 2018-02-06:
tried here as well, works as expected, so closing.

closed 2018-02-06 by nashif